pires/android-obd-reader #61: android-obd-reader crashes on Get-DTC

jeepingben, 2015-02-15:
This happens 100% of the time for me on 2 vehicles.  I have tried building the latest and using the version distributed on f-droid with the same result.

Reading live data works, but Get-DTC causes the program to crash "Unfortunately obd-reader has stopped working".  This is the exception from logcat: 
E/AndroidRuntime(26913): FATAL EXCEPTION: AsyncTask #1
E/AndroidRuntime(26913): Process: pt.lighthouselabs.obd.reader, PID: 26913
E/AndroidRuntime(26913): java.lang.RuntimeException: An error occured while executing doInBackground()
E/AndroidRuntime(26913): 	at android.os.AsyncTask$3.done(AsyncTask.java:300)
E/AndroidRuntime(26913): 	at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:355)
E/AndroidRuntime(26913): 	at java.util.concurrent.FutureTask.setException(FutureTask.java:222)
E/AndroidRuntime(26913): 	at java.util.concurrent.FutureTask.run(FutureTask.java:242)
E/AndroidRuntime(26913): 	at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:231)
E/AndroidRuntime(26913): 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
E/AndroidRuntime(26913): 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
E/AndroidRuntime(26913): 	at java.lang.Thread.run(Thread.java:841)
E/AndroidRuntime(26913): Caused by: pt.lighthouselabs.obd.exceptions.Misundersto, response: ?ption: Error running AT Z
E/AndroidRuntime(26913): 	at java.lang.Class.newInstanceImpl(Native Method)
E/AndroidRuntime(26913): 	at java.lang.Class.newInstance(Class.java:1208)
E/AndroidRuntime(26913): 	at pt.lighthouselabs.obd.commands.ObdCommand.checkForErrors(ObdCommand.java:188)
E/AndroidRuntime(26913): 	at pt.lighthouselabs.obd.commands.ObdCommand.readResult(ObdCommand.java:125)
E/AndroidRuntime(26913): 	at pt.lighthouselabs.obd.commands.ObdCommand.run(ObdCommand.java:79)
E/AndroidRuntime(26913): 	at pt.lighthouselabs.obd.reader.activity.TroubleCodesActivity$GetTroubleCodesTask.doInBackground(TroubleCodesActivity.java:196)
E/AndroidRuntime(26913): 	at pt.lighthouselabs.obd.reader.activity.TroubleCodesActivity$GetTroubleCodesTask.doInBackground(TroubleCodesActivity.java:132)
E/AndroidRuntime(26913): 	at android.os.AsyncTask$2.call(AsyncTask.java:288)
E/AndroidRuntime(26913): 	at java.util.concurrent.FutureTask.run(FutureTask.java:237)
E/AndroidRuntime(26913): 	... 4 more
W/ActivityManager( 2253):   Force finishing activity pt.lighthouselabs.obd.reader/.activity.TroubleCodesActivity
D/pt.lighthouselabs.obd.reader.activity.MainActivity(26913): Entered onStart...
D/pt.lighthouselabs.obd.reader.activity.MainActivity(26913): Resuming..
I/Timeline(26913): Timeline: Activity_idle id: android.os.BinderProxy@421ae8f0 time:82671870
W/bt-rfcomm(13170): rfc_port_closed
W/bt-btif (13170): bta_jv_rfc_port_to_cb(port_handle:0xf):jv handle:0x0 not FOUND
I/Timeline( 2253): Timeline: Activity_windows_visible id: ActivityRecord{42bb3930 u0 pt.lighthouselabs.obd.reader/.activity.MainActivity t51} time:82672140
W/SurfaceFlinger( 1724): couldn't log to binary event log: overflow.
E/WindowManager(26913): 
E/WindowManager(26913): android.view.WindowLeaked: Activity pt.lighthouselabs.obd.reader.activity.TroubleCodesActivity has leaked window com.android.internal.policy.impl.PhoneWindow$DecorView{42b8b540 V.E..... R....... 0,0-1026,642} that was originally added here
E/WindowManager(26913): 	at android.view.ViewRootImpl.<init>(ViewRootImpl.java:346)
E/WindowManager(26913): 	at android.view.WindowManagerGlobal.addView(WindowManagerGlobal.java:248)
E/WindowManager(26913): 	at android.view.WindowManagerImpl.addView(WindowManagerImpl.java:69)
E/WindowManager(26913): 	at android.app.Dialog.show(Dialog.java:286)
E/WindowManager(26913): 	at pt.lighthouselabs.obd.reader.activity.TroubleCodesActivity$GetTroubleCodesTask.onPreExecute(TroubleCodesActivity.java:153)
E/WindowManager(26913): 	at android.os.AsyncTask.executeOnExecutor(AsyncTask.java:587)
E/WindowManager(26913): 	at android.os.AsyncTask.execute(AsyncTask.java:535)
E/WindowManager(26913): 	at pt.lighthouselabs.obd.reader.activity.TroubleCodesActivity.onCreate(TroubleCodesActivity.java:76)
E/WindowManager(26913): 	at android.app.Activity.performCreate(Activity.java:5231)
E/WindowManager(26913): 	at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1087)
E/WindowManager(26913): 	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2169)
E/WindowManager(26913): 	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2271)
E/WindowManager(26913): 	at android.app.ActivityThread.access$800(ActivityThread.java:144)
E/WindowManager(26913): 	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1205)
E/WindowManager(26913): 	at android.os.Handler.dispatchMessage(Handler.java:102)
E/WindowManager(26913): 	at android.os.Looper.loop(Looper.java:136)
E/WindowManager(26913): 	at android.app.ActivityThread.main(ActivityThread.java:5146)
E/WindowManager(26913): 	at java.lang.reflect.Method.invokeNative(Native Method)
E/WindowManager(26913): 	at java.lang.reflect.Method.invoke(Method.java:515)
E/WindowManager(26913): 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:732)
E/WindowManager(26913): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:566)
E/WindowManager(26913): 	at dalvik.system.NativeStart.main(Native Method)
W/bt-l2cap(13170): L2CA_DisconnectReq()  CID: 0x0040

I see that the reset command fails in start-live-data as well, but that it keeps going.

, response: ?selabs.obd.reader.io.ObdGatewayService(30123): Failed to run command. -> Error running AT Z
D/pt.lighthouselabs.obd.reader.io.ObdGatewayService(30123): Taking job[2] from queue..
D/pt.lighthouselabs.obd.reader.io.ObdGatewayService(30123): Job state is NEW. Run it..
, response: ?selabs.obd.reader.io.ObdGatewayService(30123): Failed to run command. -> Error running AT E0
D/pt.lighthouselabs.obd.reader.io.ObdGatewayService(30123): Taking job[3] from queue..

I see that the ObdGatewayService catches all exceptions and keeps going while TroubleCodesActivity dies on exceptions (I added catch (MisunderstoodCommandException) in my build).

nomwerp, 2015-02-15:
thanks, do you actually get any useful response when collecting live data or are all the 
other commands also failing?

On Sunday 15 February 2015 13:45:00 jeepingben wrote:

This happens 100% of the time for me on 2 vehicles. I have tried building the latest and 
using the version distributed on f-droid with the same result. 
Reading live data works, but Get-DTC causes the program to crash "Unfortunately obd-
reader has stopped working". This is the exception from logcat: E/AndroidRuntime(26913): 
FATAL EXCEPTION: AsyncTask #1[1]
android.os.BinderProxy@421ae8f[2] time:82671870W/bt-rfcomm(13170): 
rfc_port_closedW/bt-btif (13170): bta_jv_rfc_port_to_cb(port_handle:0xf):jv handle:0x0 
not FOUNDI/Timeline( 2253): Timeline: Activity_windows_visible id: 
ActivityRecord{42bb3930 u0 pt.lighthouselabs.obd.reader/.activity.MainActivity t51} 
time:82672140W/SurfaceFlinger( 1724): couldn't log to binary event log: 
overflow.E/WindowManager(26913): E/WindowManager(26913): 
android.view.WindowLeaked: Activity 
pt.lighthouselabs.obd.reader.activity.TroubleCodesActivity has leaked window 
com.android.internal.policy.impl.PhoneWindow$DecorView{42b8b540 V.E..... R....... 
0,0-1026,642} that was originally added hereE/WindowManager(26913): at 
android.view.ViewRootImpl.(ViewRootImpl.java:346)E/WindowManager(26913): at 
android.view.WindowManagerGlobal.addView(WindowManagerGlobal.java:248)E/WindowMa
nager(26913): at 
android.view.WindowManagerImpl.addView(WindowManagerImpl.java:69)E/WindowManage
r(26913): at android.app.Dialog.show(Dialog.java:286)E/WindowManager(26913): at 
pt.lighthouselabs.obd.reader.activity.TroubleCodesActivity$GetTroubleCodesTask.onPreEx
ecute(TroubleCodesActivity.java:153)E/WindowManager(26913): at 
android.os.AsyncTask.executeOnExecutor(AsyncTask.java:587)E/WindowManager(26913): 
at android.os.AsyncTask.execute(AsyncTask.java:535)E/WindowManager(26913): at 
pt.lighthouselabs.obd.reader.activity.TroubleCodesActivity.onCreate(TroubleCodesActivity.
java:76)E/WindowManager(26913): at 
android.app.Activity.performCreate(Activity.java:5231)E/WindowManager(26913): at 
android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1087)E/WindowMa
nager(26913): at 
android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2169)E/WindowMan
ager(26913): at 
android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2271)E/WindowMana
ger(26913): at 
android.app.ActivityThread.access$800(ActivityThread.java:144)E/WindowManager(26913)
: at 
android.app.ActivityThread$H.handleMessage(ActivityThread.java:1205)E/WindowManager
(26913): at 
android.os.Handler.dispatchMessage(Handler.java:102)E/WindowManager(26913): at 
android.os.Looper.loop(Looper.java:136)E/WindowManager(26913): at 
android.app.ActivityThread.main(ActivityThread.java:5146)E/WindowManager(26913): at 
java.lang.reflect.Method.invokeNative(Native Method)E/WindowManager(26913): at 
java.lang.reflect.Method.invoke(Method.java:515)E/WindowManager(26913): at 
com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:732)E/Windo
wManager(26913): at 
com.android.internal.os.ZygoteInit.main(ZygoteInit.java:566)E/WindowManager(26913): at 
dalvik.system.NativeStart.main(Native Method)W/bt-l2cap(13170): L2CA_DisconnectReq() 
CID: 0x0040 
I see that the reset command fails in start-live-data as well, but that it keeps going. 
, response: ?selabs.obd.reader.io.ObdGatewayService(30123): Failed to run command. -> 
Error running AT ZD/pt.lighthouselabs.obd.reader.io.ObdGatewayService(30123): Taking 
job[2] from queue..D/pt.lighthouselabs.obd.reader.io.ObdGatewayService(30123): Job 
state is NEW. Run it.., response: ?selabs.obd.reader.io.ObdGatewayService(30123): Failed to 
run command. -> Error running AT 
E0D/pt.lighthouselabs.obd.reader.io.ObdGatewayService(30123): Taking job[3] from 
queue.. 
I see that the ObdGatewayService catches all exceptions and keeps going while 
TroubleCodesActivity dies on exceptions (I added catch 
(MisunderstoodCommandException) in my build). 

--------
[1] https://github.com/pires/android-obd-reader/issues/1
[2] https://github.com/android.os.BinderProxy/android-obd-reader/commit/421ae8f0
[3] https://github.com/pires/android-obd-reader/issues/61
[4] 
https://github.com/notifications/beacon/AJpBTvVUXTl9T6Gn_4XKL1Cm00IZIZ2Iks5nsQrc
gaJpZM4DgvE6.gif

jeepingben, 2015-02-15:
Live data gives correct results on both vehicles.  RPM and coolant temperature are believable. No values had garbage.

jeepingben, 2015-02-25:
Does it make sense to migrate the trouble-codes activity to the same more async looking command processor that live-data uses?  It's been a few days since I looked in the code, but I don't remember a reason that it wouldn't work.  If you think it makes sense, I could take a look at it (no promise it will be done in a timely manner)

nomwerp, 2015-02-25:
yes, it would make a lot of sense if live-data and trouble code activity share more of the code since they effectively do the same thing...   but on the other hand if we can prevent it from crashing by catching one more exception we should probably do that first.

nomwerp, 2015-03-11:
should be fixed by #63 , we should create a separate issue for improving trouble code activity

closed 2015-03-11 by nomwerp