pires/android-obd-reader #67: Enable/disable bluetooth only for use

elagin, 2015-03-09:
I create pull request https://github.com/pires/android-obd-reader/pull/66
Should bluetooth connection be turned on and off  into startLiveData() and stopLiveData() ?

pires, 2015-03-09:
Well, it's an interesting idea to ask the user to enable it before starting the app and helping him doing that. But of course, we still need to validate the device is paired as well.

We can also ask the user if he wants to disconnect Bluetooth when the app is stopped.

elagin, 2015-03-09:
The idea was to store the state of bluetooth before running the application and return it after the call stopLiveData ()
It is necessary to check whether the fast and reliable work if turn on BT device and immediately start reading data.

elagin, 2015-03-09:
I try call btAdapter.enable() info startLiveData(). Data acquisition process starts after recall startLiveData() into menu.
Possibly some function invoked asynchronously when the connection has not yet occurred .

pires, 2015-03-10:
> The idea was to store the state of bluetooth before running the application and return it after the call stopLiveData ()

What do you mean by `store the state of bluetooth`?

I believe thinks work as expected now and we only need to validate more scenarios when Bluetooth is enabled/disabled by the user.

elagin, 2015-03-10:
In MainActivity.onCreate call BluetoothAdapter.isEnabled() and store result info variable.
Then, for example in MainActivityonDestroy(), if ConfigActivity.ENABLE_BT_KEY && !variable call call BluetoothAdapter.Disable()

In the setting of only one item "Enable Bluetooth"

pires, 2015-03-10:
Oh, now I get it. Well, it's good to try turning it on/off depending on the app lifecycle. I'll be waiting for a PR to better assess this behavior.

nomwerp, 2015-03-10:
interesting, i was attempting something similar but instead of using

 `btAdapter.enable();`

i was trying something like:

`Intent enableBtIntent = new Intent(BluetoothAdapter.ACTION_REQUEST_ENABLE);`
`startActivityForResult(enableBtIntent, ENABLE_BLUETOOTH);`

and some more code to check if it was successful

` public void onActivityResult(int requestCode, int resultCode, Intent data) {
    switch (requestCode) {
      case ENABLE_BLUETOOTH:
        if (resultCode != RESULT_OK) {
          showDialog(BLUETOOTH_DISABLED);
          Toast.makeText(this, "Bluetooth is disabled, will use Mock service instead", Toast.LENGTH_SHORT).show();
        }
    }
  }`

would that give any benefit at all or just complicate things?

pires, 2015-03-10:
Intents are always the best way to do it in Android.

nomwerp, 2015-03-11:
one scenario that seems to be missed is to turn bluetooth off if it was automatically enabled  + app is paused + no live data is running ?  

1) bluetooth is off 
2) app started
3) bluetooth gets enabled
4) live data is started
5) live data is stopped
6) app paused
7) should bluetooth be turned off again?

elagin, 2015-03-22:
If turn on bluetooth after start live data on my device i have trouble.
This is probably due to the fact that the connections necessary for some time, and this algorithm code which would have a connection in the event of repeated failure. Of course it makes sense to on bluetooth for use only (between start and stop).

pires, 2015-05-10:
Can we close this one?

nomwerp, 2015-05-10:
yes works great, my previous comment about disabling bluetooth when paused can be ignored, i don't think it should do that anyway and bluetooth does get disabled onDestroy().

closed 2015-05-10 by nomwerp