pires/android-obd-reader #69: add accelerometer data

nomwerp, 2015-03-11:
It would be great to have access to accelerometer data in the app (and especially future database uploads) , for example to correct for fuel usage for driving uphill or correlate engine load with power etc.

pires, 2015-05-10:
I wish we could just get proper fuel consumption support.

Pho3niX90, 2015-07-24:
Accelerometer wont be needed for proper fuel consumption. I have already made pull requests. The method will be to see how much fuel is actually injected and then calculate the fuel being burned. Most vehicles do have trims to actually show how the afr is being changed (example. Going uphil, going WOT ect)

pires, 2015-07-24:
We'll see how that works out :)

closed 2015-12-08 by pires