pires/android-obd-reader #72: Bluetooth is disabled after closing the app

FunThomas76, 2015-03-19:
Hi, 
normally my bluetooth is always on but after closing this app my bluetooth is turned off :(

best regards,
Thomas

pires, 2015-03-19:
@elagin @nomwerp I believe this was added recently, right?

elagin, 2015-03-19:
Yes recently been added. Now try to fix.

pires, 2015-03-19:
Thanks @elagin. Waiting for a PR then ;-)

elagin, 2015-03-19:
Ready  #73

pires, 2015-03-19:
@FunThomas76 fix LGTM. Care to try it out and let us know if it worked?

FunThomas76, 2015-03-23:
i could not believe that you could fix it that fast ;-) 
i have to rebuild today evening and will report you then

pires, 2015-03-23:
@FunThomas76 reopening and waiting on your feedback to close the issue ;-)

FunThomas76, 2015-03-23:
working as expected :)
after closing the app my bluetooth stays enabled like always
Thanx

closed 2015-03-23 by FunThomas76

pires, 2015-03-23:
Awesome!

commit f0231bc5 (2015-03-19): Merge pull request #73 from elagin/bluetooth_off_error

Fixes #72