pires/android-obd-reader #78: Definition of available PIDs

elagin, 2015-03-25:
OBD2 is support definition of available PIDs. Command 0100 - get avaible PIDs from 1 to 20.
For example on my Suzuki Sx4 (2008) return: BE 3F B8 13
10111110001111111011100000010011

But in the current implementation ObdCommand not possible to use only supported PIDs. It would be great to make a list of command PIDs and use it when creating commands. Such as is make in FuelTrim enum.

pires, 2015-06-08:
I can't understand. Can you please improve issue description?

pires, 2015-07-09:
Implemented `obd-java-api` support in 0.10, https://github.com/pires/obd-java-api/blob/e2a98f051407414edff5001097887446cfad8106/src/main/java/pt/lighthouselabs/obd/commands/protocol/AvailablePidsObdCommand.java.

Not having more details on how to implement this in this project, I'm closing this.

closed 2015-07-09 by pires