pires/android-obd-reader #80: New UI based on Material Design

pires, 2015-05-09:


pires, 2015-05-09:
Need help!

nomwerp, 2015-05-09:
I guess it could make things pretty but it seems that android:Theme.Material requires API level 21 (current min is 10). since supposedly only 10% of devices use this is that something we want? or does it fall back gracefully? (although it might not install anymore)
https://developer.android.com/about/dashboards/index.html

nomwerp, 2015-05-09:
to answer my own question:
from main/res/values/styles.xml
"Theme customizations available in newer API levels can go in
            res/values-vXX/styles.xml, while customizations related to
            backward-compatibility can go here."

nomwerp, 2015-05-09:
just applying the theme and deleting the background colours in favour of the defaults doesn't even look too bad: 

[image]

pires, 2015-05-09:
I guess it never looked as prettier :D

nomwerp, 2015-05-09:
commit 8bebec600e  (still  looks as ugly as always on older versions )

but I'll play around for a bit with the new theme...

pires, 2015-05-10:
Awesome! Thanks

nomwerp, 2015-05-10:
Theme.Material.Light looks better in my opinion: 

[image]

nomwerp/android-obd-reader@0a8bc8c806

I'm having trouble getting the fancy shadows to show up , setting elevation works in the preview but not on device

fernando-costa, 2015-05-10:
i believe that you can expand compatibility using Android Support Library

http://developer.android.com/tools/support-library/index.html

pires, 2015-05-14:
@nomwerp we're ready to close this one, are we not?

closed 2015-05-15 by pires