pires/android-obd-reader #82: New Icons

nomwerp, 2015-05-14:
we should have a icon!

with some hints from http://www.google.com/design/spec/style/icons.html#icons-product-icons
i've made an attempt at drawing a svg (here attached as 144X144):  

[image]

basic sprites from https://github.com/google/material-design-icons " released under an Attribution 4.0 International license." but it seems that the "Bluetooth SIG" also claims it as a trademark http://www.bluetooth.com/Pages/Trademarks.aspx  

I welcome harsh criticism on any aspect, the idea, colours , style, etc.

pires, 2015-05-14:
Oh, so nice of you to actually get a new icon. It looks good :)

nomwerp, 2015-05-14:
where would be a good place to put .svg?  my best bet is directly in src/main/res

pires, 2015-05-14:
I have no idea.

pires, 2015-05-14:
Did you fix this with #83? If so, next time close it please.

nomwerp, 2015-05-14:
well if no complaints about the new icons i'd be happy to close...

closed 2015-05-14 by nomwerp

pires, 2015-05-14:
You merged it and were waiting on complains? :-D