pires/android-obd-reader #85: Auto Protocol selection maybe not working

anti43, 2015-05-15:
Hello,

I can connect with torque etc well, however if uisng android-odb-reader I cannot get past the protocol selection message.

pt.lighthouselabs.obd.exceptions.MisunderstoodCommandException: Error running AT E0, response: +CONNECTING<<A0:F?
	at java.lang.Class.newInstanceImpl(Native Method)
	at java.lang.Class.newInstance(Class.java:1319)
	at pt.lighthouselabs.obd.commands.ObdCommand.checkForErrors(ObdCommand.java:206)
	at pt.lighthouselabs.obd.commands.ObdCommand.readResult(ObdCommand.java:137)
	at pt.lighthouselabs.obd.commands.ObdCommand.run(ObdCommand.java:84)
	at io.ObdGatewayService.executeQueue(ObdGatewayService.java:207)
	at io.AbstractGatewayService$1.run(AbstractGatewayService.java:40)
	at java.lang.Thread.run(Thread.java:864)

Do you have any idea what goes wrong here?

Many thanks

Andreas

nomwerp, 2015-05-15:
hmmm it could be that the "CONNECTING"  in the response is causing some confusion , had  a similar issue with device returning "SEARCHING..."  in front of the first response.

if that is the case a crude fix could be to strip it from the response in  https://github.com/pires/obd-java-api/blob/master/src/main/java/pt/lighthouselabs/obd/commands/ObdCommand.java at the end of readRawData() add something like: 
    `rawData = rawData.replace("SEARCHING...", "");`

anti43, 2015-05-15:
Ok thanks I'll analyze the junk data..

pires, 2015-05-15:
I'm open to improvements on the https://github.com/pires/obd-java-api project.

anti43, 2015-05-15:
ok I did a fork and will do pull requests when i found the issue!

pires, 2015-05-15:
Awesome @anti43!

anti43, 2015-05-16:
Ok,

I did some debugging now with my car and an obd terminal..
it seems there is the sequence +CONNECTING<<A0:F? >NNECTED ? at the beginning of the communication .. some sort of initializing message. plus the searching.. message is on front of any command response, not only on error codes..

pires, 2015-05-16:
Different cars and ELM327 devices respond differently. It's sad but it's the reality. I'm open to any improvement that adds support for responses like this without breaking previous functionality.

anti43, 2015-05-16:
Ok,

reissung the rest command as you do with hte second echo command does solve the issue so nothing to chnage in the obd-api part.
- but then mostly i get some "unable to connect". I wonder how torque manages to do this as there ist working like always:)

nomwerp, 2015-05-16:
try with only a couple of OBD commands selected in the preferences, i've found that if you ask for one that is not supported you might be left with "unable to connect" until the next reset

nomwerp, 2015-05-16:
it seems that for me having the (additional) "Reset OBD" enabled in settings results in "UNABLE TO CONNECT" for all following commands

anti43, 2015-05-16:
yes you are absolutely right. possibly issuing a reset every 4 seconds is killing the elm :)

nomwerp, 2015-05-16:
ok just tested it and it is indeed the extra reset command that screws things up, I had it disabled in my preferences because one of the commands was causing issues but hadn't narrowed down to the extra reset.  i think the issue is that calling reset without the other init commands leaves it in a non working state.  anyway i removed if from the list of commands to run : commit e1a7a8776dcecafec2e69fb939f7c8e472911f65

but back to your original issue , did you solve that ?

anti43, 2015-05-16:
Well, yes and no. Just ignoring it and running a dummy command and ignoring the outcome as very first thing makes any troubles go away.

nomwerp, 2015-05-16:
without the extra reset that stopped everything doesn't it just fail on the first command (there is a temperature command hard coding in for this purpose just after all the init stuff) and continue happily after that?

closed 2015-06-08 by pires