pires/android-obd-reader #86: 2.0RC

nomwerp, 2015-05-18:
@pires how about releasing a 2.0RC and push it to f-droid too?

or are there any show stoppers that we need to address first?

pires, 2015-05-18:
@nomwerp would be nice to close some [obj-java api issues](https://github.com/pires/obd-java-api/issues), release 1.0 and update the dependency here as well, but I'm lacking the time to do it.

nomwerp, 2015-05-19:
alright, i'll have a look but there don't seem to be (m)any open issues that affect android-obd-reader.  

I think it'd also be good to do a bit more work on the UI before calling it 2.0

f-droid doesn't seem to be auto updating anyway , last build there is from 2013, what needs to happen to get that to work again?

pires, 2015-05-19:
@nomwerp honestly, I never pushed the app to any app store. Probably, someone else did it. I've been notified about it in the past.

nomwerp, 2015-05-20:
ok i'll investigate , seems that we'd need to make an update of : 
https://gitlab.com/fdroid/fdroiddata/blob/master/metadata/eu.lighthouselabs.obd.reader.txt
I might create a 2.0-expermintal or so to test if f-droid can actually build it from source

nomwerp, 2015-05-20:
managed to have a local copy of f-droid build it from source successfully, only thing we need to add to f-droid metadata is another build:

```
Build:2.0,11
    commit=826dd9737
    gradle=yes
```
...where commit is now just the latest commit but can also be a tag when we actually release a 2.0

there is one issue: at f-droid the app is listed as eu.lighthouselabs.obd.reader instead of pt.lighthouselabs.obd.reader so we'd have to add it as a new app or possibly rename the existing, or maybe tell f-droid to ignore this issue

pires, 2015-05-20:
@nomwerp you are the expert here :D I'll leave that to you. I will be tagging RC1 in the coming days.

nomwerp, 2015-05-20:
do you know the history behind the name change?

pires, 2015-05-20:
@nomwerp I know I've changed the packaging in the past because I no longer controlled the `lighthouselabs.eu` domain.  But right now, I don't even control the `.pt` as well, so I can probably rename stuff to `com.github.pires.obdreader` or something prettier.

pires, 2015-06-08:
Moved to `com.github.pires.obd` in 26cba74.

Tagging `2.0-RC1` now.

closed 2015-06-08 by pires

nomwerp, 2015-06-08:
OK great,   when I'm back home I'll make a pull request for the fdroid people with the name change and to set it to auto update based on the tags.