pires/android-obd-reader #87: automatically start collecting the data

vivek40, 2015-05-20:
Hi,

I had gone through your source code, am developing one such application. All i want to know is how efficient is to implement a feature like when the user goes out of the Bluetooth range and comes back again to the range, if the service is running the background it should automatically start collecting the data without intimating the user. 

How feasible is the requirement, else if you guide me how to go about on it, i would try implementing it.

pires, 2015-05-20:
It is feasible. One would need to handle Bluetooth errors on https://github.com/pires/android-obd-reader/blob/master/src/main/java/pt/lighthouselabs/obd/reader/io/ObdGatewayService.java and play with the service lifecycle to act accordingly.

vivek40, 2015-05-20:
So its basically when even the connectivity is lost, (i.e) in ObdGatewayService.java, inside executeQueue(), catch block will be called. But if i call the method startService() at that time, some times AT Z command fails and comes into the catch block. So i should ignore this condition for AT commands alone?

pires, 2015-05-20:
This is most probably due to [the explicit delay in sending commands](https://github.com/pires/obd-java-api/blob/master/src/main/java/pt/lighthouselabs/obd/commands/ObdCommand.java#L111).

pires, 2015-06-08:
@vivek40 is there a PR coming?

closed 2015-12-08 by pires