pires/android-obd-reader #88: Problem with Troubles Codes

oneal, 2015-05-20:
I want to obtains troubles codes of my ELM327, i'm using your api for OBD, but this library doesnt want to  give me the errors, this said me command OBD failure, I try to charge the application obd reader of your repository and i have the same problem. I think the aplication fail in this part     

                    ModifiedTroubleCodesObdCommand tcoc = new ModifiedTroubleCodesObdCommand();
                    tcoc.run(sock.getInputStream(), sock.getOutputStream());
                    result = tcoc.getFormattedResult();

but i don't undertand because.

pires, 2015-05-20:
Need logs please.

oneal, 2015-05-20:
05-21 00:12:34.363    6079-7238/pt.lighthouselabs.obd.obd2 W/System.err﹕ pt.lighthouselabs.obd.exceptions.UnableToConnectException: Error running 03, response: SEARCHING...
UNABLETOCONNECT
05-21 00:12:34.364    6079-7238/pt.lighthouselabs.obd.obd2 W/System.err﹕ at java.lang.Class.newInstanceImpl(Native Method)
05-21 00:12:34.364    6079-7238/pt.lighthouselabs.obd.obd2 W/System.err﹕ at java.lang.Class.newInstance(Class.java:1319)
05-21 00:12:34.367    6079-7238/pt.lighthouselabs.obd.obd2 W/System.err﹕ at pt.lighthouselabs.obd.commands.ObdCommand.checkForErrors(ObdCommand.java:211)
05-21 00:12:34.367    6079-7238/pt.lighthouselabs.obd.obd2 W/System.err﹕ at pt.lighthouselabs.obd.commands.ObdCommand.readResult(ObdCommand.java:140)
05-21 00:12:34.367    6079-7238/pt.lighthouselabs.obd.obd2 W/System.err﹕ at pt.lighthouselabs.obd.commands.ObdCommand.run(ObdCommand.java:87)
05-21 00:12:34.367    6079-7238/pt.lighthouselabs.obd.obd2 W/System.err﹕ at pt.lighthouselabs.obd.obd2.TroubleCodes$GetTroubleCodesTask.doInBackground(TroubleCodes.java:210)
05-21 00:12:34.368    6079-7238/pt.lighthouselabs.obd.obd2 W/System.err﹕ at pt.lighthouselabs.obd.obd2.TroubleCodes$GetTroubleCodesTask.doInBackground(TroubleCodes.java:132)
05-21 00:12:34.368    6079-7238/pt.lighthouselabs.obd.obd2 W/System.err﹕ at android.os.AsyncTask$2.call(AsyncTask.java:287)
05-21 00:12:34.368    6079-7238/pt.lighthouselabs.obd.obd2 W/System.err﹕ at java.util.concurrent.FutureTask.run(FutureTask.java:234)
05-21 00:12:34.369    6079-7238/pt.lighthouselabs.obd.obd2 W/System.err﹕ at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:230)
05-21 00:12:34.370    6079-7238/pt.lighthouselabs.obd.obd2 W/System.err﹕ at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1080)
05-21 00:12:34.370    6079-7238/pt.lighthouselabs.obd.obd2 W/System.err﹕ at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:573)
05-21 00:12:34.371    6079-7238/pt.lighthouselabs.obd.obd2 W/System.err﹕ at java.lang.Thread.run(Thread.java:838)

pires, 2015-05-20:
You are not connected.

oneal, 2015-05-21:
ELM321 is connected, this occure in your application android-obd-rearder too.

pires, 2015-05-21:
> pt.lighthouselabs.obd.exceptions.UnableToConnectException: Error running 03, response: SEARCHING...UNABLETOCONNECT

This response comes from the ELM327 device.

nomwerp, 2015-05-21:
i think @pires means that the  *ELM327* is unable to connect to the car? e.g. the car is not responding because it is not running/powered  or the appropriate init and reset commands haven't been issued

pires, 2015-05-21:
Exactly! Thank you @nomwerp

oneal, 2015-05-21:
ah!!! okay thanks!! should i change ELM 327 by other??

nomwerp, 2015-05-21:
did you try it with the engine running?

oneal, 2015-05-21:
he said me no data when i try to engine running

oneal, 2015-05-22:
sorry, it's send me a lot of informar ion with engine running

closed 2015-05-22 by pires