pires/android-obd-reader #90: Add full internationalization support

pires, 2015-05-22:


nomwerp, 2015-05-31:
i think that's all strings in the UI, @amouly would you mind to do some more translation to spanish?

amouly, 2015-06-01:
@nomwerp yes, I added a lot more Strings, but I have that issue with indentation, because I made some changes and can't merge directly. Maybe I can discard my changes on indentation and just pull the Strings.

nomwerp, 2015-06-01:
yeah i'm afraid that the changes i did in  19fa27811ebb0b19147f4a2e077334689d820706 are causing the conflict,  could i ask you to prepare a pull request just for the changes you made in src/main/res/values-es/strings.xml  and  src/main/res/values/strings.xml ?  and I'll work work out what those strings are called now afterwards.  thanks.

pires, 2015-06-15:
@nomwerp @amouly can you update me on this, please?

amouly, 2015-06-15:
I think is mostly done. Right @nomwerp?

pires, 2015-06-15:
Awesome!

closed 2015-06-15 by pires