pires/android-obd-reader #93: Max obd polling rate

brianharms, 2015-05-31:
Hi Pires, thanks for your work on such a great app. Forgive me if this is not the right place to post this issue, but I've been having trouble getting data faster than about 5-6 times per second.

I have only engine rpm selected, and I've modified the project slightly to accept milliseconds for the period (instead of just seconds). It seems to be accurate when polling at 1000 ms, 500ms, and even 250 or 200 ms. But I can't seem to get any more than

brianharms, 2015-05-31:
That. I'm wondering if there is a max polling rate set by the modules themselves (I have a Bluetooth elm27 module) or if there is some overhead in the app I can trim out to get faster polling rates. If the latter is the case, any suggestions are welcome. 

Best,
brian

nomwerp, 2015-05-31:
in  obd-java-api/src/main/java/pt/lighthouselabs/obd/commands/ObdCommand.java

```java
    /*
     * HACK GOLDEN HAMMER ahead!!
     * 
     * Due to the time that some systems may take to respond, let's give it
     * 200ms.
     */
    Thread.sleep(200);
  ```

brianharms, 2015-05-31:
Excellent, thanks for the response, nomwerp.

Should I then download the zipped obd-java-api project, add it as a library to the obd reader app (while removing the existing external dependency, whose files are locked) and edit it there? 

Best,
brian

nomwerp, 2015-05-31:
you can clone obd-java-api it into your project folder:
`git clone https://github.com/pires/obd-java-api.git`

create `obd-java-api/build.gradle` with something like
`apply plugin: 'java'`

in main build.gradle change 
 `compile 'pt.lighthouselabs.obd:obd-java-api:0.8'` 
to
 `compile project(':obd-java-api')`

in settings.gradle add `include ':obd-java-api'`

that should do it

brianharms, 2015-05-31:
Thanks for the quick response!

closed 2015-05-31 by brianharms