pires/android-obd-reader #94: Imperial/Metric Units Option

Hippoish24, 2015-06-07:
###### Summary
The preference to enable imperial units does not currently change anything.

###### Proposed change
Add the following method to `pt.lighthouselabs.obd.reader.io.ObdGatewayService`:

```java
  @Override
  public void queueJob(ObdCommandJob job){
    // This is a good place to enforce the imperial units option
    job.getCommand().useImperialUnits(prefs.getBoolean(ConfigActivity.IMPERIAL_UNITS_KEY, false));

    // Now we can pass it along
    super.queueJob(job);
  }
```

###### Rationale
When values are retrieved from an ObcCommand, it uses the getFormattedResult() method.  This method contains logic for printing either imperial or metric units, but depends on the invocation of the useImperialUnits(boolean) method to change this value; by default, it is false.  All commands that are created and issued are wrapped into ObdCommandJobs and dispatched using the queueJob(...) method in ObdGatewayService (inherited from AbstractGatewayService).  Since ObdGatewayService also has access to the SharedPreferences, it makes sense to make this change here.

An alternate solution would be to inject the preferences into the AbstractGatewayService and enforce the option to use imperial units in the existing implementation of queueJob(...), though that doesn't seem as neat from a design perspective.

pires, 2015-06-07:
Care to fill a PR? Will gladly review.

Hippoish24, 2015-06-07:
Will do!

Hippoish24, 2015-06-07:
Fixed as of pull #95.  Thank you! :)

closed 2015-06-07 by Hippoish24

pires, 2015-06-07:
Thank you!! You're awesome.