pires/android-obd-reader #96: Gradle Wrapper

amouly, 2015-06-07:
We should include the Gradle Wrapper to the root directory of the project. This is an standard way to import the project on Android Studio, and not recreating this files on each machine.

pires, 2015-06-07:
I've never had any issues importing this project into Android Studio. Actually, Gradle Wrapper used to exist for people that didn't have Gradle installed which I don't believe is true when you install said IDE.

amouly, 2015-06-07:
No, Android Stusio only have the Gradle Plugin, but not Gradle itself. I
haven't got it installed, I just have AS and it creates the wrapper for
each project. It's a common issue when you import a project, not having
Gradle or the Wrapper and nothing works :P.

pires, 2015-06-07:
So if Android Studio creates the wrapper, what's the issue then?

amouly, 2015-06-07:
No, AS just created it when you start the project from 0, but not when you
import it. When so, AS ask where the wrapper is, or where Gradle is, but if
you haven't got anyone, you can't run the project. The solution is copy the
wrapper from another project and paste it on the current project.

pires, 2015-06-07:
Hmmm, current Android Studio complains about not having Gradle indeed. I wonder if they removed the bundled version... anyway one of the prerequisites in the doc is `Gradle 2.2.1`. Will mark this as an improvement rather than a bug and will take care of it ASAP. Thank you for reporting.

amouly, 2015-06-07:
Yeah sure. Maybe if you commit and push the wrapper it just works. If you
want to add it to another branch, I can test importing the project from 0.
Thanks.

pires, 2015-06-07:
Actually, I revert what I just said and yes, Gradle is bundled.

```
$ ls /Applications/Android\ Studio.app/Contents/gradle/gradle-2.2.1/bin/gradle*
/Applications/Android Studio.app/Contents/gradle/gradle-2.2.1/bin/gradle
/Applications/Android Studio.app/Contents/gradle/gradle-2.2.1/bin/gradle.bat
```

pires, 2015-06-07:
PR on its way.

amouly, 2015-06-08:
@pires the wrapper is working just fine. But i'm getting an error when executing the app (compiling and running on real device). The stack is the following:

06-08 20:54:19.948    1633-1633/? E/AndroidRuntime﹕ FATAL EXCEPTION: main
    Process: com.github.pires.obd.reader, PID: 1633
    java.lang.RuntimeException: Unable to start activity ComponentInfo{com.github.pires.obd.reader/com.github.pires.obd.reader.activity.MainActivity}: java.lang.IllegalStateException: Unable to use annotation database(s)
            at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2325)
            at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2387)
            at android.app.ActivityThread.access$800(ActivityThread.java:151)
            at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1303)
            at android.os.Handler.dispatchMessage(Handler.java:102)
            at android.os.Looper.loop(Looper.java:135)
            at android.app.ActivityThread.main(ActivityThread.java:5254)
            at java.lang.reflect.Method.invoke(Native Method)
            at java.lang.reflect.Method.invoke(Method.java:372)
            at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:903)
            at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:698)
     Caused by: java.lang.IllegalStateException: Unable to use annotation database(s)
            at roboguice.RoboGuice.initializeAnnotationDatabaseFinderAndHierarchyTraversalFilterFactory(RoboGuice.java:292)
            at roboguice.RoboGuice.getOrCreateBaseApplicationInjector(RoboGuice.java:149)
            at roboguice.RoboGuice.getOrCreateBaseApplicationInjector(RoboGuice.java:90)
            at roboguice.RoboGuice.getInjector(RoboGuice.java:197)
            at roboguice.activity.RoboActivity.onCreate(RoboActivity.java:90)
            at com.github.pires.obd.reader.activity.MainActivity.onCreate(MainActivity.java:305)
            at android.app.Activity.performCreate(Activity.java:5990)
            at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1106)
            at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2278)
            at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2387)
            at android.app.ActivityThread.access$800(ActivityThread.java:151)
            at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1303)
            at android.os.Handler.dispatchMessage(Handler.java:102)
            at android.os.Looper.loop(Looper.java:135)
            at android.app.ActivityThread.main(ActivityThread.java:5254)
            at java.lang.reflect.Method.invoke(Native Method)
            at java.lang.reflect.Method.invoke(Method.java:372)
            at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:903)
            at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:698)
     Caused by: java.lang.RuntimeException: java.lang.ClassNotFoundException: AnnotationDatabaseImpl
            at com.google.inject.AnnotationDatabaseFinder.<init>(AnnotationDatabaseFinder.java:47)
            at com.google.inject.Guice.setAnnotationDatabasePackageNames(Guice.java:133)
            at roboguice.RoboGuice.initializeAnnotationDatabaseFinderAndHierarchyTraversalFilterFactory(RoboGuice.java:288)
            at roboguice.RoboGuice.getOrCreateBaseApplicationInjector(RoboGuice.java:149)
            at roboguice.RoboGuice.getOrCreateBaseApplicationInjector(RoboGuice.java:90)
            at roboguice.RoboGuice.getInjector(RoboGuice.java:197)
            at roboguice.activity.RoboActivity.onCreate(RoboActivity.java:90)
            at com.github.pires.obd.reader.activity.MainActivity.onCreate(MainActivity.java:305)
            at android.app.Activity.performCreate(Activity.java:5990)
            at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1106)
            at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2278)
            at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2387)
            at android.app.ActivityThread.access$800(ActivityThread.java:151)
            at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1303)
            at android.os.Handler.dispatchMessage(Handler.java:102)
            at android.os.Looper.loop(Looper.java:135)
            at android.app.ActivityThread.main(ActivityThread.java:5254)
            at java.lang.reflect.Method.invoke(Native Method)
            at java.lang.reflect.Method.invoke(Method.java:372)
            at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:903)
            at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:698)
     Caused by: java.lang.ClassNotFoundException: AnnotationDatabaseImpl
            at java.lang.Class.classForName(Native Method)
            at java.lang.Class.forName(Class.java:309)
            at java.lang.Class.forName(Class.java:273)
            at com.google.inject.AnnotationDatabaseFinder.getAnnotationDatabaseInstance(AnnotationDatabaseFinder.java:73)
            at com.google.inject.AnnotationDatabaseFinder.<init>(AnnotationDatabaseFinder.java:36)
            at com.google.inject.Guice.setAnnotationDatabasePackageNames(Guice.java:133)
            at roboguice.RoboGuice.initializeAnnotationDatabaseFinderAndHierarchyTraversalFilterFactory(RoboGuice.java:288)
[... cut ...]
            at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:698)
    Suppressed: java.lang.ClassNotFoundException: AnnotationDatabaseImpl
            at java.lang.Class.classForName(Native Method)
            at java.lang.BootClassLoader.findClass(ClassLoader.java:781)
            at java.lang.BootClassLoader.loadClass(ClassLoader.java:841)
            at java.lang.ClassLoader.loadClass(ClassLoader.java:504)
            ... 26 more
     Caused by: java.lang.NoClassDefFoundError: Class not found using the boot class loader; no stack available

amouly, 2015-06-09:
@pires I don't think that this issue it's related to the wrapper, if anyone else have the same issue maybe we can open a new one for this.

pires, 2015-06-09:
@nomwerp I think it's related to `proguard` blocking `roboguice` stuff. Will look into it, thanks.

amouly, 2015-06-09:
@pires this is fixed.

PR #97 (2015-06-08): Added Gradle wrapper files.
Fixes #96
commit e49365e2 (2015-06-08): Added Gradle wrapper files. Fixes #96
closed 2015-06-09 by amouly