pires/android-obd-reader #102: 2.0-RC1 f-droid build crashed on start

nomwerp, 2015-06-20:


nomwerp, 2015-06-20:
D/VoldCmdListener( 210): cryptfs getpwtype D/roboguice.RoboGuice(11500): Using annotation database(s). D/roboguice.RoboGuice(11500): Using annotation database(s) : [, roboguice] W/System.err(11500): java.lang.ClassNotFoundException: AnnotationDatabaseImpl W/System.err(11500): at java.lang.Class.classForName(Native Method) W/System.err(11500): at java.lang.Class.forName(Class.java:308) W/System.err(11500): at java.lang.Class.forName(Class.java:272) W/System.err(11500): at com.google.inject.AnnotationDatabaseFinder.getAnnotationDatabaseInstance(AnnotationDatabaseFinder.java:73) W/System.err(11500): at com.google.inject.AnnotationDatabaseFinder.<init>(AnnotationDatabaseFinder.java:36) W/System.err(11500): at com.google.inject.Guice.setAnnotationDatabasePackageNames(Guice.java:133) W/System.err(11500): at roboguice.RoboGuice.initializeAnnotationDatabaseFinderAndHierarchyTraversalFilterFactory(RoboGuice.java:288) W/System.err(11500): at roboguice.RoboGuice.getOrCreateBaseApplicationInjector(RoboGuice.java:149) W/System.err(11500): at roboguice.RoboGuice.getOrCreateBaseApplicationInjector(RoboGuice.java:90) W/System.err(11500): at roboguice.RoboGuice.getInjector(RoboGuice.java:197) W/System.err(11500): at roboguice.activity.RoboActivity.onCreate(RoboActivity.java:90) W/System.err(11500): at com.github.pires.obd.reader.activity.MainActivity.onCreate(MainActivity.java:305) W/System.err(11500): at android.app.Activity.performCreate(Activity.java:5933) W/System.err(11500): at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1105) W/System.err(11500): at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2282) W/System.err(11500): at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2389) W/System.err(11500): at android.app.ActivityThread.access$900(ActivityThread.java:147) W/System.err(11500): at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1296) W/System.err(11500): at android.os.Handler.dispatchMessage(Handler.java:102) W/System.err(11500): at android.os.Looper.loop(Looper.java:135) W/System.err(11500): at android.app.ActivityThread.main(ActivityThread.java:5254) W/System.err(11500): at java.lang.reflect.Method.invoke(Native Method) W/System.err(11500): at java.lang.reflect.Method.invoke(Method.java:372) W/System.err(11500): at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:898) W/System.err(11500): at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:693) W/System.err(11500): Caused by: java.lang.ClassNotFoundException: Didn't find class "AnnotationDatabaseImpl" on path: DexPathList[[zip file "/data/app/com.github.pires.obd.reader-1/base.apk"],nativeLibraryDirectories=[/vendor/lib, /system/lib]] W/System.err(11500): at dalvik.system.BaseDexClassLoader.findClass(BaseDexClassLoader.java:56) W/System.err(11500): at java.lang.ClassLoader.loadClass(ClassLoader.java:511) W/System.err(11500): at java.lang.ClassLoader.loadClass(ClassLoader.java:469) W/System.err(11500): ... 25 more W/System.err(11500): Suppressed: java.lang.ClassNotFoundException: AnnotationDatabaseImpl W/System.err(11500): at java.lang.Class.classForName(Native Method) W/System.err(11500): at java.lang.BootClassLoader.findClass(ClassLoader.java:781) W/System.err(11500): at java.lang.BootClassLoader.loadClass(ClassLoader.java:841) W/System.err(11500): at java.lang.ClassLoader.loadClass(ClassLoader.java:504) W/System.err(11500): ... 26 more W/System.err(11500): Caused by: java.lang.NoClassDefFoundError: Class not found using the boot class loader; no stack available D/AndroidRuntime(11500): Shutting down VM D/VoldCmdListener( 210): cryptfs getpwtype E/AndroidRuntime(11500): FATAL EXCEPTION: main E/AndroidRuntime(11500): Process: com.github.pires.obd.reader, PID: 11500 E/AndroidRuntime(11500): java.lang.RuntimeException: Unable to start activity ComponentInfo{com.github.pires.obd.reader/com.github.pires.obd.reader.activity.MainActivity}: java.lang.IllegalStateException: Unable to use annotation database(s) E/AndroidRuntime(11500): at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2329) E/AndroidRuntime(11500): at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2389) E/AndroidRuntime(11500): at android.app.ActivityThread.access$900(ActivityThread.java:147) E/AndroidRuntime(11500): at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1296) E/AndroidRuntime(11500): at android.os.Handler.dispatchMessage(Handler.java:102) E/AndroidRuntime(11500): at android.os.Looper.loop(Looper.java:135) E/AndroidRuntime(11500): at android.app.ActivityThread.main(ActivityThread.java:5254) E/AndroidRuntime(11500): at java.lang.reflect.Method.invoke(Native Method) E/AndroidRuntime(11500): at java.lang.reflect.Method.invoke(Method.java:372) E/AndroidRuntime(11500): at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:898) E/AndroidRuntime(11500): at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:693) E/AndroidRuntime(11500): Caused by: java.lang.IllegalStateException: Unable to use annotation database(s) E/AndroidRuntime(11500): at roboguice.RoboGuice.initializeAnnotationDatabaseFinderAndHierarchyTraversalFilterFactory(RoboGuice.java:292) E/AndroidRuntime(11500): at roboguice.RoboGuice.getOrCreateBaseApplicationInjector(RoboGuice.java:149) E/AndroidRuntime(11500): at roboguice.RoboGuice.getOrCreateBaseApplicationInjector(RoboGuice.java:90) E/AndroidRuntime(11500): at roboguice [... cut ...]  at java.lang.BootClassLoader.findClass(ClassLoader.java:781) E/AndroidRuntime(11500): at java.lang.BootClassLoader.loadClass(ClassLoader.java:841) E/AndroidRuntime(11500): at java.lang.ClassLoader.loadClass(ClassLoader.java:504) E/AndroidRuntime(11500): ... 26 more E/AndroidRuntime(11500): Caused by: java.lang.NoClassDefFoundError: Class not found using the boot class loader; no stack available W/ActivityManager( 1956): Force finishing activity com.github.pires.obd.reader/.activity.MainActivity

amouly, 2015-06-20:
@nomwerp ProGuard problem again maybe?

nomwerp, 2015-06-20:
yes maybe... this is the 2.0-RC1 tag from 12 days ago so wouldn't have included #100

nomwerp, 2015-06-20:
here is the log with line ends ;)
```
D/roboguice.RoboGuice(11500): Using annotation database(s).
D/roboguice.RoboGuice(11500): Using annotation database(s) : [, roboguice]
W/System.err(11500): java.lang.ClassNotFoundException: AnnotationDatabaseImpl
W/System.err(11500):    at java.lang.Class.classForName(Native Method)
W/System.err(11500):    at java.lang.Class.forName(Class.java:308)
W/System.err(11500):    at java.lang.Class.forName(Class.java:272)
W/System.err(11500):    at com.google.inject.AnnotationDatabaseFinder.getAnnotationDatabaseInstance(AnnotationDatabaseFinder.java:73)
W/System.err(11500):    at com.google.inject.AnnotationDatabaseFinder.<init>(AnnotationDatabaseFinder.java:36)
W/System.err(11500):    at com.google.inject.Guice.setAnnotationDatabasePackageNames(Guice.java:133)
W/System.err(11500):    at roboguice.RoboGuice.initializeAnnotationDatabaseFinderAndHierarchyTraversalFilterFactory(RoboGuice.java:288)
W/System.err(11500):    at roboguice.RoboGuice.getOrCreateBaseApplicationInjector(RoboGuice.java:149)
W/System.err(11500):    at roboguice.RoboGuice.getOrCreateBaseApplicationInjector(RoboGuice.java:90)
W/System.err(11500):    at roboguice.RoboGuice.getInjector(RoboGuice.java:197)
W/System.err(11500):    at roboguice.activity.RoboActivity.onCreate(RoboActivity.java:90)
W/System.err(11500):    at com.github.pires.obd.reader.activity.MainActivity.onCreate(MainActivity.java:305)
W/System.err(11500):    at android.app.Activity.performCreate(Activity.java:5933)
W/System.err(11500):    at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1105)
W/System.err(11500):    at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2282)
W/System.err(11500):    at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2389)
W/System.err(11500):    at android.app.ActivityThread.access$900(ActivityThread.java:147)
W/System.err(11500):    at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1296)
W/System.err(11500):    at android.os.Handler.dispatchMessage(Handler.java:102)
W/System.err(11500):    at android.os.Looper.loop(Looper.java:135)
W/System.err(11500):    at android.app.ActivityThread.main(ActivityThread.java:5254)
W/System.err(11500):    at java.lang.reflect.Method.invoke(Native Method)
W/System.err(11500):    at java.lang.reflect.Method.invoke(Method.java:372)
W/System.err(11500):    at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:898)
W/System.err(11500):    at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:693)
W/System.err(11500): Caused by: java.lang.ClassNotFoundException: Didn't find class "AnnotationDatabaseImpl" on path: DexPathList[[zip file "/data/app/com.github.pires.obd.reader-1/base.apk"],nativeLibraryDirectories=[/vendor/lib, /system/lib]]
W/System.err(11500):    at dalvik.system.BaseDexClassLoader.findClass(BaseDexClassLoader.java:56)
W/System.err(11500):    at java.lang.ClassLoader.loadClass(ClassLoader.java:511)
W/System.err(11500):    at java.lang.ClassLoader.loadClass(ClassLoader.java:469)
W/System.err(11500):    ... 25 more
W/System.err(11500):    Suppressed: java.lang.ClassNotFoundException: AnnotationDatabaseImpl
W/System.err(11500):            at java.lang.Class.classForName(Native Method)
W/System.err(11500):            at java.lang.BootClassLoader.findClass(ClassLoader.java:781)
W/System.err(11500):            at java.lang.BootClassLoader.loadClass(ClassLoader.java:841)
W/System.err(11500):            at java.lang.ClassLoader.loadClass(ClassLoader.java:504)
W/System.err(11500):            ... 26 more
W/System.err(11500):    Caused by: java.lang.NoClassDefFoundError: Class not found using the boot class loader; no stack available
D/AndroidRuntime(11500): Shutting down VM
D/VoldCmdListener(  210): cryptfs getpwtype
E/AndroidRuntime(11500): FATAL EXCEPTION: main
E/AndroidRuntime(11500): Process: com.github.pires.obd.reader, PID: 11500
E/AndroidRuntime(11500): java.lang.RuntimeException: Unable to start activity ComponentInfo{com.github.pires.obd.reader/com.github.pires.obd.reader.activity.MainActivity}: java.lang.IllegalStateException: Unable to use annotation database(s)
E/AndroidRuntime(11500):        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2329)
E/AndroidRuntime(11500):        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2389)
E/AndroidRuntime(11500):        at android.app.ActivityThread.access$900(ActivityThread.java:147)
E/AndroidRuntime(11500):        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1296)
E/AndroidRuntime(11500):        at android.os.Handler.dispatchMessage(Handler.java:102)
E/AndroidRuntime(11500):        at android.os.Looper.loop(Looper.java:135)
E/AndroidRuntime(11500):        at android.app.ActivityThread.main(ActivityThread.java:5254)
E/AndroidRuntime(11500):        at java.lang.reflect.Method.invoke(Native Method)
E/AndroidRuntime(11500):        at java.lang.reflect.Method.invoke(Method.java:372)
E/AndroidRuntime(11500):        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:898)
E/AndroidRuntime(11500):        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:693)
E/AndroidRuntime(11500): Caused by: java.lang.IllegalStateException: Unable to use annotation database(s)
E/AndroidRuntime(11500):        at roboguice.RoboGuice.initializeAnnotationDatabaseFinderAndHierarchyTraversalFilterFactory(RoboGuice.java:292)
[... cut ...]
E/AndroidRuntime(11500):                at java.lang.BootClassLoader.findClass(ClassLoader.java:781)
E/AndroidRuntime(11500):                at java.lang.BootClassLoader.loadClass(ClassLoader.java:841)
E/AndroidRuntime(11500):                at java.lang.ClassLoader.loadClass(ClassLoader.java:504)
E/AndroidRuntime(11500):                ... 26 more
E/AndroidRuntime(11500):        Caused by: java.lang.NoClassDefFoundError: Class not found using the boot class loader; no stack available
W/ActivityManager( 1956):   Force finishing activity com.github.pires.obd.reader/.activity.MainActivity
```

nomwerp, 2015-06-20:
but it sounds like we need to tag 2.0-rc2 to get that proguard fix included

nomwerp, 2015-06-21:
Version 2.0-RC2 has now been build by f-droid and is available https://f-droid.org/repository/browse/?fdid=com.github.pires.obd.reader

closed 2015-06-21 by nomwerp