pires/android-obd-reader #104: Polling and receiving data is indredibly slow

Pho3niX90, 2015-06-23:
I have tested the app as is but have removed the delau of 200ms in script and also changed the polling frequency to .3 seconds. It still takes about 3secs to poll, and in those 3 secs a lot of information is lost.

Pho3niX90, 2015-06-23:
Just as a note.

Compared to Torque, torque polls double the sensors in half the time

pires, 2015-06-23:
Where did you remove the 200ms? I'm not sure I follow what `in script` means. And polling time.

pires, 2015-06-23:
And please, stop comparing this to Torque. Torque is a closed app, paid by many users. This **is not**!

closed 2015-06-23 by pires

Pho3niX90, 2015-06-23:
Excuse me but your troubleshooting method in the readme states to compare and test in torque as well. 

Code removed was

    /*
     * HACK GOLDEN HAMMER ahead!!
     * 
     * Due to the time that some systems may take to respond, let's give it
     * 200ms.
     */
    Thread.sleep(200);

pires, 2015-06-23:
`Thread.sleep(...)` happens in `obd-java-api` as well, as a safe-measure. Change it, compile it and adapt your app at your own risk. It's the only way to bypass said safe-measure.

Regarding Torque, it's a troubleshooting thing for misbehaving ELM 327 devices, not for comparing it to this open-source application.