pires/android-obd-reader #107: Settings for the polling rate Or adaptive polling

MaxBenChrist, 2015-07-10:
It would be nice to be able to either set the update period in 

```java

public static int getObdUpdatePeriod(SharedPreferences prefs) {
    String periodString = prefs
        .getString(ConfigActivity.OBD_UPDATE_PERIOD_KEY, "4"); // 4 as in seconds
    int period = 4000; // by default 4000ms
```

a) to a user selected value in the configure menu (maybe in the range of 1-6 seconds)
or 
b) make the polling adaptive by a caller/listener combination to achieve the fastest possible rate.

For example my obd and car combination is able to send data every 1.5-2 seconds.

nomwerp, 2015-07-17:
sorry if i'm missing something but shouldn't it already be taken from the settings, or it is broken?

```java
  public static int getObdUpdatePeriod(SharedPreferences prefs) {
    String periodString = prefs.
            getString(ConfigActivity.OBD_UPDATE_PERIOD_KEY, "4"); // 4 as in seconds
    int period = 4000; // by default 4000ms

    try {
      period = (int) (Double.parseDouble(periodString) * 1000);
    } catch (Exception e) {
    }

    if (period <= 0) {
      period = 4000;
    }

    return period;
  }
```

MaxBenChrist, 2015-07-18:
Yes, the settings are working! So a) is solved. My mistake. 
I am not sure how to implement 

Maybe we should close this issue?

closed 2015-07-18 by pires