pires/android-obd-reader #108: Test obd-java-api 1.0-RC2

pires, 2015-07-11:
@nomwerp @amouly care to help?

nomwerp, 2015-07-13:
builds and runs but getting  "7F 01 12" as response from several commands which i guess is the new UnsupportedCommandException?

nomwerp, 2015-07-13:
we should also add the new commands to https://github.com/pires/android-obd-reader/blob/master/src/main/java/com/github/pires/obd/reader/config/ObdConfig.java so that they become available

nomwerp, 2015-07-13:
to answer my own question:

> getting "7F 01 12" as response from several commands which i guess is the new  UnsupportedCommandException?

yes it is , since it is now actually throwing an exception the raw response is shown. we should probably make it more pretty by replacing  "7F 01 12" with "NA" or "-"  or something like that (and possibly maybe disabling unsupported command for remaining of the live data collection cycle?)

pires, 2015-07-13:
@nomwerp want to take care of that? Would be happy to merge. I'm traveling for a couple weeks and my availability for this is close to none.

nomwerp, 2015-07-14:
yes will have a look , in any case I'll merge the pull request and raise issue for adding new commands and prettifying unsupported command results

pires, 2015-07-15:
Can you test RC3. Just released it but may take a while to sync to Maven Central.

nomwerp, 2015-07-15:
commented on pires/obd-java-api#34

pires, 2015-07-15:
Thank you. Will close this and wait for @anti43 feedback on #34.

PR #109 (2015-07-14): Moved to com.github.pires:obd-java-api:1.0-RC2.
Fixes #108
commit a82b39cf (2015-07-14): Moved to com.github.pires:obd-java-api:1.0-RC2.
Fixes #108
closed 2015-07-15 by pires