pires/android-obd-reader #111: Unable to get DTC

mvglasow, 2015-07-14:
I have recently tried the latest version available on F-Droid. Unlike in the earlier version I tried before, I can now obtain live data on Android 4.4. However, when I try to get DTCs, I still get an error about the adapter not being reachable and am unable to get any information from the dongle. Could this be a leftover from #7? That is, live data addresses the OBD dongle in a way that is compatible with Android 4.2.1+ but DTC still uses the old way?

pires, 2015-07-15:
@nomwerp any insight?

nomwerp, 2015-07-15:
no not really without diving into it, works for me on Android 5-something...   any suggestions welcome

pires, 2015-08-02:
@mvglasow can you try latest code, please?

mvglasow, 2015-12-13:
Tried today with 2.0-RC2, still no luck. OS is now Android 5.1.1.

pires, 2015-12-13:
What's your car protocol?

mvglasow, 2015-12-13:
Where can I find that out? The car is a 2005 Audi A4 (B7 model).

pires, 2015-12-13:
Have you tried building the code?

mvglasow, 2015-12-13:
No, so far I've only used the APKs on F-Droid.

mvglasow, 2015-12-13:
A question, though: how come live data works while DTC doesn't? I thought the underlying protocol is the same for both – or am I wrong?

pires, 2015-12-13:
You should try build and install the code. There's protocol nuances for some commands.

pires, 2015-12-14:
@valens254 anything to add?

valens254, 2015-12-14:
@mvglasow I can't understand, what you get when trying read DTC

pires, 2015-12-14:
@valens254 just to be sure it's clear, @mvglasow is running an old release of this software. I will try and make a new release, but I suggested him to compile the code.

Also @mvglasow, I'm not responsible for F-Droid releases or any other APK releases.

valens254, 2015-12-14:
Yes, the old release can read all except DTC and VIN code, because not all protocols was implemented yet.

pires, 2015-12-14:
@mvglasow please test https://github.com/pires/android-obd-reader/releases/tag/2.0-RC3 and let us know. I'm closing this for now.

mvglasow, 2015-12-20:
Same symptoms here, too: I can collect live data but when I try to get DTC, it says "Unable to connect to Bluetooth device". The error message indicates the error happens lower down in the stack – or the app is giving out a misleading error message.

As for F-Droid: for most apps they simply build an APK from the repo. Thus, apart from the signature, there is no difference between an F-Droid build, a version I have built myself and (presumably) a release build on github. They may occasionally patch apps to remove advertising or non-free dependencies, but they try to limit this to a few exceptional cases because they lack the resources to do such adaptations.

pires, 2015-12-20:
@mvglasow did you run the RC3?

mvglasow, 2015-12-20:
Yes, downloaded from https://github.com/pires/android-obd-reader/releases/download/2.0-RC3/android-obd-reader-debug.apk. The app shows as RC2 in the package manager, but that is presumably due to https://github.com/pires/android-obd-reader/blob/2.0-RC3/src/main/AndroidManifest.xml#L6.

pires, 2015-12-21:
You are correct. @mvglasow the app crashed? Can you provide logs?

Pho3niX90, 2015-12-21:
I have fixed this issue in my version. The issue was that it crashes when there are no errors found. I will see if I can incorporate my fix with this.

pires, 2015-12-21:
@pho3nix90 care to develop further your comment? Is it an app thing? Library thing? I can provide a fix as well if I see some logs.

Pho3niX90, 2015-12-21:
@pires is an app issue. I will provide a fix in the next 30mins.

pires, 2015-12-21:
@pho3nix90 awesome! Thank you.

Pho3niX90, 2015-12-21:
Try this: https://github.com/pires/android-obd-reader/pull/153

pires, 2015-12-21:
@mvglasow please try https://github.com/pires/android-obd-reader/releases/tag/2.0-RC4

closed 2015-12-21 by pires

mvglasow, 2015-12-21:
@pires OK, will try. Just to clarify: the app does not crash, it simply displays a toast with an error message.

mvglasow, 2015-12-21:
Same issue with RC4. When I try to read DTC, all I get is a toast saying "unable to connect to Bluetooth device".

Logcat is attached. Since at the same time Android was syncing my phone book and dumping all its contents to the log, I've filtered for `com.github.pires`. If you need other data, let me know.

[alogcat.2015-12-21-15-22-32+0100.filtered.txt](https://github.com/pires/android-obd-reader/files/68570/alogcat.2015-12-21-15-22-32.0100.filtered.txt)

Pho3niX90, 2015-12-21:
@mvglasow are you able to read any other data? Have you conpletely restrted your phone and then ONLY tried reading dtc without clicking on start live data?

mvglasow, 2015-12-21:
@pires I have a suspicion what it might be: LiveData and DTC define their own methods to connect to the OBD dongle, and implementations differ.

Compare ObdGatewayService (live data):
https://github.com/pires/android-obd-reader/blob/2.0-RC4/src/main/java/com/github/pires/obd/reader/io/ObdGatewayService.java#L138

and DTC:
https://github.com/pires/android-obd-reader/blob/2.0-RC4/src/main/java/com/github/pires/obd/reader/activity/TroubleCodesActivity.java#L295

ObdGatewayService has a workaround for certain Bluetooth connection issues, which is missing for DTC. Unfortunately my logcat got cut off at the start, so I can't tell for sure if live data relies on the fallback on my phone, but it's probably a good idea to implement the fallback in DTC as well.

mvglasow, 2015-12-21:
@Pho3niX90 see above – yes, live data works. Didn't try restarting, though. My test cycle was:

* Upgrade
* Start live data – works.
* Stop live data.
* Get DTC – error.
* Start live data – works.
* Stop live data.
* Get DTC – error.

Currently my suspicion is that it's a low-level issue and that my phone requires the hack in ObdGatewayService to get anything out of the dongle at all, and DTC won't work because it's lacking the fallback machanism.

pires, 2015-12-21:
Interesting!

Pho3niX90, 2015-12-21:
Try #155, that should work if it is in fact your issue

pires, 2015-12-21:
@mvglasow care to try #155?

mvglasow, 2015-12-21:
I've just built it myself and implemented a fix, which is functionally equivalent to #155. In fact it works now. I'll just clean up the code to make it a bit more modular (duplicate code isn't very nice) and then send you a merge request.

pires, 2015-12-21:
@mvglasow i'm also in favor if a dedup of code.

mvglasow, 2015-12-21:
Just built and tested successfully (live data and DTC work).

I've introduced a new class, io.BluetoothManager, to take the new method, because I didn't find any other class that looked appropriate for this. If you have a better idea (you certainly know the code better than I do), feel free to refactor.

pires, 2015-12-21:
@mvglasow this was a really nice contribution. Thank you for that. And thank you for debugging this issue until a solution was found.

nomwerp, 2016-01-11:
i set up F-droid to automatically build from github but that works on the  android:versionName 
in AndroidManifest.xml ..  so if someone updates that to match the latest tag  (exactly!) it should 
automagically appear on f-droid 

On Sunday 20 December 2015 08:52:25 mvglasow wrote:

Same symptoms here, too: I can collect live data but when I try to get DTC, it says "Unable to 
connect to Bluetooth device". The error message indicates the error happens lower down in the 
stack – or the app is giving out a misleading error message. 
As for F-Droid: for most apps they simply build an APK from the repo. Thus, apart from the 
signature, there is no difference between an F-Droid build, a version I have built myself and 
(presumably) a release build on github. They may occasionally patch apps to remove advertising 
or non-free dependencies, but they try to limit this to a few exceptional cases because they lack 
the resources to do such adaptations. 

--------
[1] https://github.com/pires/android-obd-reader/issues/111#issuecomment-166134871
[2] https://github.com/notifications/beacon/
AJpBTtXf97GYlddrtgixrKxUrz2ayh3mks5pRtRJgaJpZM4FYQXB.gif

PR #154 (2015-12-21): Fixes #111
Fixes #111
Fixes #153
commit 4e8ee3d6 (2015-12-21): Merge pull request #154 from pires/153-fix_111

Fixes #111
Fixes #153
commit 81abaa94 (2015-12-21): Fixes #111