pires/android-obd-reader #112: show more friendly response for unsupported commands

nomwerp, 2015-07-14:
from  #108 : 

since move to obd-java-api 1.0-RC2 i'm getting "7F 01 12" as response from several commands which is due to the new UnsupportedCommandException  , since it is now actually throwing an exception the raw response is shown. we should make it more pretty by replacing "7F 01 12" with "NA" or "-" or something like that (and possibly maybe disabling unsupported command for remaining of the live data collection cycle?)

nomwerp, 2015-07-15:
with obd-java-api 1.0-RC3 it now returns "7F0112" instead of "7F 01 12" ...

pires, 2015-07-15:
Good point here. The buffer processing stuff on the API changed a bit and it's a bit confusing, so I'd say that for now we should change how this app handles strings. I'm sorry for breaking this, but I can't pay as much attention as I want for the next two weeks :(

commit fa1e60d5 (2015-07-14): fixes #112 , replace "7F 01 12" in raw response with NA (Not Available) , don't update obd status field .
commit 0094257d (2015-07-17): fixes #112 again, this time by catching unsupported command exceptions
closed 2015-07-17 by nomwerp