pires/android-obd-reader #119: VinObdCommand issue

Pho3niX90, 2015-07-20:
Hi.

I tried out the vin and seems that it returns a 70 digit number instead of a vin number. Ex "4902010000005749020242534548490203393230334902043043473949020539383331"

nomwerp, 2015-07-20:
thangs for testing, i believe this had just been fixed in pires/obd-java-api#50

Pho3niX90, 2015-07-20:
Awesome. I will test that and revert back

Pho3niX90, 2015-07-20:
This is working like a charm now.

pires, 2015-07-20:
I'll release `obd-java-api` RC5 later today and upgrade the dependency version here.

Pho3niX90, 2015-07-21:
Could we please reopen this? I tried it on my 750 today and it couldn't properly pick up the vin, instead it returned "IWIBLI82I61I1".

nomwerp, 2015-07-21:
is there a difference between the version you tested an RC5?

pires, 2015-07-21:
See https://github.com/pires/obd-java-api/commit/d5474dce0556d3eca1707a5b41033edbc0600c94. That should be the properly formatted VIN.

pires, 2015-07-21:
Calling @anti43 on this one.

anti43, 2015-07-22:
@nomwerp yes with this fix it works properly for me. one way to screw it is if the obd device is sending headers, which you can turn off issuing a command AT H0 which is not yet implemented in RC5. Had this issue yesterday.

Pho3niX90, 2015-07-22:
Yes I am using the correct version. This is working on the one car but not the other.

anti43, 2015-07-22:
when you connect via obd terminal, what is the raw response?

Pho3niX90, 2015-07-22:
I haven't tested. Which obd terminal do you suggest

anti43, 2015-07-22:
obd terminal from play store

Pho3niX90, 2015-07-23:
I added the headers command and that solved it. Thank you.

Pho3niX90, 2015-07-30:
Receiving this on an M3 = 0140:4902015742531:574439323031302:50593335393632

anti43, 2015-07-30:
see https://github.com/pires/obd-java-api/issues/54

Pho3niX90, 2015-07-30:
Is there a possible solution to this? This -> 4902015742531:574439323031302:50593335393632
is the vin

anti43, 2015-07-30:
yes there is a solution, see the referenced issue. the presented solution is not ideal as discussed, but if it works for you it might be ok

Pho3niX90, 2015-07-30:
That solution just yields hex code returned

anti43, 2015-07-30:
then go ahead and use it

```
 StringBuilder b = new StringBuilder();

 for (int i = 2; i < buffer.size(); i++) {
      b.append(new Character((char) buffer.get(i).intValue()).toString());
  }
            
 String vin = b.toString().replaceAll("[\u0000-\u001f]", "");
```

See a complete solution here (parses those ":" separated vins and teh other multiline one):

https://github.com/anti43/obd-java-api/blob/master/src/main/java/com/github/pires/obd/commands/control/VinObdCommand.java

Pho3niX90, 2015-07-31:
I have implemented your solution, vin now comes back IIBSI92I0JI68

Pho3niX90, 2015-07-31:
Please ignore the above. Just noticed this is on an E46 m3. I am testing the app on a fleet of BMWs

anti43, 2015-07-31:
can you post the complete raw response

Pho3niX90, 2015-09-08:
So it has creeped up again with another car. IWIBCI38I05I5

pires, 2015-12-08:
Please, try the latest code. Closing this but will reopen if there's feedback about it not being fixed.

closed 2015-12-08 by pires