pires/android-obd-reader #120: Better fuel consumption

Pho3niX90, 2015-07-20:
Hi.

In my car (bmw m6) the app can't get the fuel consumption via obd. I have found a website that gives a proper calculation for this use. 

This is the same method most of the OBD2 apps use on the playstore. 

Source: http://www.lightner.net/obd2guru/IMAP_AFcalc.html
Making use of the Ideal Gas Law, a synthetic variable called IMAP can be used to estimate the mass air-flow (MAF) of an internal combustion vehicle, much like a MAF sensor. However, in order to make this work, the engine displacement and volumetric efficiency of the engine must be provided.

IMAP is integrated by the firmware using the following formula:

        IMAP = RPM * MAP / IAT / 2
where RPM is engine speed in revolutions per minute, MAP (manifold absolute pressure) is measured in KPa, and IAT (intake air temperature) is measured in degrees Kelvin. This integrated value can be converted into total air flow (grams) using the following formula:
        (grams of air) = (IMAP/60)*(Vol Eff/100)*(Eng Disp)*(MM Air)/(R)
where R is 8.314 J/°K/mole and the average molecular mass of air is 28.97 g/mole. Note that in the above formula the volumetric efficiency of the engine is measured in percent and the displacement is in liters.
Once we have MAF integrated over time, we can calculate fuel flow using a specific air/fuel mixing ratio (A/F). A/F can range from 6 to 22. The total mass of air which flows through the engine (as measured by the MAF sensor or calculated using MAP, RPM and IAT) is divided by the A/F ratio (e.g., 14.7) to derive the total mass of fuel that has been burned.

If the air mass is measured in grams, then the calculated amount of fuel will also be in grams. Using an average density of gasoline of 6.17 pounds per gallon and knowing that there are 454 grams per pound, then dividing the total fuel mass in grams by (454 * 6.17) yields the total number of gallons of fuel burned for a given amount of air (as measured by the MAF sensor).

Fuel consumed can be estimated from integrated MAF using the following formula:

        (gallons of fuel) = (grams of air) / (air/fuel ratio) / 6.17 / 454
Knowing the total fuel consumed and the distance traveled, the fuel consumption rate in MPG (miles per gallon) then can be calculated.

pires, 2015-07-20:
I've tried a lot of equations before and never worked out because engines are so different. I'm open to try out an implementation, though.

Pho3niX90, 2015-07-21:
I have started implementing this into my version. Sadly I can only test with the car since obdsim only allows limited readings. If all goes according to plan I should have this working this week.

pires, 2015-07-21:
@Pho3niX90 awesome!

Pho3niX90, 2015-07-24:
So I have started this. Need to test on multiple vehicles though first.
For the time being I have made the pull request for everyone to test the 2 different AFRs as it is crucial to the consumption
https://github.com/pires/obd-java-api/pull/55

pires, 2015-07-24:
@Pho3niX90 when adding new commands to `obd-java-api`, please add more tests as well, please. I will be releasing 1.0-RC6 in a bit.

Pho3niX90, 2015-07-24:
@pires Which tests are you referring to?

pires, 2015-07-24:
@Pho3niX90 unit-tests. Check `obd-java-api/src/test/java`.

pires, 2015-08-02:
@Pho3niX90 can we close this?

closed 2015-12-08 by pires