pires/android-obd-reader #123: VinObdCommand get vin only once per session

Pho3niX90, 2015-07-20:
Since the vin will never change on a vehicle, I think a improvement will be to only get the vin once at the start of a logging session. If this is currently the case I apologize for creating this.

This will bring down the overhead on polling.

nomwerp, 2015-07-20:
i think this applies in general , for example a unsupported command in unlikely to suddenly become supported. so maybe if a command return same value x number of times don't request it again in that session ?  but that might not be a good idea for all commands

Pho3niX90, 2015-07-20:
Yes that would be the ideal solution. There is a class that checks the supported PIDs, when a session is started it would be good to run this first and store the PID's with the vin, so next time it knows what can be polled and what can't.

nomwerp, 2015-07-20:
or maybe move both the preference dialogue, i.e. before (only show available) or after(check selected) selecting the commands.  and for vin, since we already have a setting for vin there we could add option to get it from the car.

Pho3niX90, 2015-07-21:
I will try to add the feature, I am craving for faster data

pires, 2015-07-21:
@Pho3niX90 you're rocking the place! Keep it up.

closed 2015-12-08 by pires