pires/android-obd-reader #124: PHP Server version

Pho3niX90, 2015-07-20:
I have created a php version of the upload server. Not sure how I can contribute it to the project?

nomwerp, 2015-07-20:
yep did the same thing ;) not sure where to put it either

Pho3niX90, 2015-07-22:
Here you go. https://github.com/Pho3niX90/OBD2-SERVER

Its extremely simple, just add credentials and it will automatically create the database and add the new columns, if new readings gets added it will automatically create the new columns as well.

pires, 2015-08-02:
@Pho3niX90 is that a _server_? Been a while since I coded PHP but I do not see any REST related code but rather a PHP script that one can run in his terminal.

closed 2015-08-02 by pires

Pho3niX90, 2015-09-08:
@pires yes it is a server. That's the end url, have been running it. 

The brilliant part about it is that it automatically creates the tables and fields as required, so maintenance on it is null

carpiko1010, 2017-07-16:
I am using this php code on my server, but there is no entries in the mysql table, the tables are created but no data is getting entered into it from the obd app.