pires/android-obd-reader #127: Missing @string/action_clear_codes?

Hippoish24, 2015-07-23:
What does ```@string/action_clear_codes``` in [trouble_codes.xml](https://github.com/pires/android-obd-reader/blob/master/src/main/res/menu/trouble_codes.xml) reference?  It's not in [strings.xml](https://github.com/pires/android-obd-reader/blob/master/src/main/res/values/strings.xml), and neither myself nor Android Studio can find it elsewhere.  As a result, Gradle is unable to build the project.

Pho3niX90, 2015-07-23:
It's this:

    <string name="action_clear_codes">Clear Codes</string>

And obdreadings fix is 

https://github.com/pires/android-obd-reader/pull/125/files

nomwerp, 2015-07-23:
@Pho3niX90 I merged #128 for a quick fix while we sort out the code formatting

Pho3niX90, 2015-07-23:
@nomwerp okay thats perfect

Hippoish24, 2015-07-23:
Looks good; thank you!

closed 2015-07-23 by Hippoish24