pires/android-obd-reader #130: Thoughts on moving away from Roboguice

Pho3niX90, 2015-07-24:
What are your thoughts?

I just personally think the less unnecessary library's that are used, the better the app will respond. Native android code has better performance

pires, 2015-07-24:
I will not remove roboguice in favor of more boilerplate code.

Hippoish24, 2015-07-24:
What responsiveness would removing Roboguice add?  As I understand it, it's only used when initializing the app for creating the UI elements.

pires, 2015-07-24:
@Hippoish24 it's also helping a lot with service and `AsyncTask` lifecycle management. Again, this is a no-go for me unless something happens to how much Android code we need to handle this burden.

closed 2015-07-24 by pires

Pho3niX90, 2015-07-24:
@Hippoish24 http://developer.android.com/training/articles/memory.html#DependencyInjection

If you really need to use a framework I would say use http://google.github.io/dagger/

amouly, 2015-07-24:
+1 for Dagger.

pires, 2015-07-24:
Dagger is much more complicated to use and I want to maintain the app simple. I don't see any added value for the user in this, since the application delivers what it's supposed to do.

Still, I'm open to analyze a PR that replaces Roboguice with something that means more performance **AND** less code.