pires/android-obd-reader #131: Need guideline/help

Pho3niX90, 2015-07-28:
I am using a different obd2 reader to test commands on more cars, but I am having an issue where all info returned is incorrect.

For example please see attached image
What could the cause be?
[image]

Pho3niX90, 2015-07-28:
The car was stationary. So non of those readings are correct.

Pho3niX90, 2015-07-28:
Here is a screenshot from terminal

[image]

MaxBenChrist, 2015-07-28:
You tried to restart the app? I mean by closing it the app settings in the general android menu?

I observed a similar behavior when I stopped the engine during a trip and the bt connection to the doongle was lost. The application was not able to connect again and when I started a new trip (even with a stationary car) it read 80 km/h and 167 rpm all the time.

For me killing the application in the menu and starting it again helped.

pires, 2015-07-28:
This is probably caused by the service not being able to reset/restart.

MaxBenChrist, 2015-07-28:
Yeah, I was thinking the same. You got any idea how to fix that properly? Maybe the trip should be finished if for a certain number of readings no values were observed?

Pho3niX90, 2015-07-29:
Fixed it.

pires, 2015-07-29:
How was it fixed?

Pho3niX90, 2015-07-29:
Seems that it isn't fixed. What I can gather is that for some dongles echo doesn't always turn off.

sytolk, 2015-08-14:
I have the same issue with my toyota corolla verso
[image]

results is the same if car is moving or not. I have try to restart application restart Android phone but without success. @Pho3niX90 what is this terminal ?

sytolk, 2015-09-09:
09-09 16:43:20.520 E/com.github.pires.obd.reader.io.ObdGatewayService( 3354): Failed to run command. -> Error running 09 02, response: 0902NODATA

sytolk, 2015-09-16:
I have update obd-api lib to version 10. And test this issue with cheap but working with other obd app bluetooth hardware. The only changes are speed from 20km/h to 65 km/h but the real speed =0km/h
Its have trouble code true but the code is different..
[image]

Pho3niX90, 2015-09-16:
@sytolk You are receiving echos. Some adapters require you to turn on echo and then off twice. Give that a try.

Pho3niX90, 2015-09-16:
@sytolk also make sure headers are off

sytolk, 2015-09-16:
I will try this. Thanks @Pho3niX90 But why these commands are not executed from obd-reader app?
https://www.scantool.net/support/index.php?_m=knowledgebase&_a=viewarticle&kbarticleid=32
ATE0 - Turn echo off. Characters sent to ElmScan are not retransmitted back to the host computer.
ATE1 - Turn echo on. This is the default state, characters are echoed back to the host computer.
ATH0 - Turn headers off. This is the default state, header information and CRC byte are omitted.
ATH1 - Turn headers on. Header information and CRC byte are displayed.
ATZ - Causes the chip to reset itself, as if the power was cycled off and then on again.

Pho3niX90, 2015-09-16:
@sytolk they are sent to turn of by default. But I have noticed on some dongles they need to be turned on and then off to correctly work.

pires, 2015-12-08:
@Pho3niX90 care to add latest comment to documentation?

commit d59a3d14 (2015-12-23): Fixes #131
commit 1d1861d4 (2015-12-23): Fixes #131
Fixes #145
closed 2015-12-23 by pires