pires/android-obd-reader #132: http method for upload should be POST

matthiasblaesing, 2015-07-28:
When I first used android-obd-reader I wrote a simple rest backend, wait for POSTs. From the settings and from my understanding of http methods this would have been ok. It turns out PUT requests are issued instead.

While it would be more backwards compatible to just change the settings description I think it would be wrong. From RFC 2616 [http://www.w3.org/Protocols/rfc2616/rfc2616-sec9.html]:

For POST:
> [...] The POST method is used to request that the origin server accept the entity enclosed in the request as a new subordinate of the resource identified by the Request-URI in the Request-Line. POST is designed to allow a uniform method to cover the following functions: [...]
>  - Extending a database through an append operation.
> [...]

For PUT:
> The PUT method requests that the enclosed entity be stored under the supplied Request-URI. If the Request-URI refers to an already existing resource, the enclosed entity SHOULD be considered as a modified version of the one residing on the origin server.

My interpretation: PUT updates an existing entity or creates a new entity with a known identity, while the POST is a generic "here you have a new entity, handle it" request.

In this case it is a new dataset, that has no identity, until the target server assigns one. So I'd vote to change OdbService to indicate a POST operation.

pires, 2015-07-28:
@matthiasblaesing very good insight actually. It's been a long time since I last worked on that code and at the time I didn't gave it much thought. I should have :+1:

I'll be working on this.

pires, 2015-08-02:
@nomwerp @anti43 @MaxBenChrist @Pho3niX90 just letting you know (since you are the most active users around) that `PUT` has been replaced with `POST`.

closed 2015-08-02 by pires

MaxBenChrist, 2015-08-04:
Thanks for the notification!

I implemented a multi entity post in my branch. If somebody is interested, here is the snippet:

``` java
HttpClient client = new DefaultHttpClient();
        // TODO: option einführen
        HttpPost post = new HttpPost( );

        MultipartEntityBuilder builder = MultipartEntityBuilder.create();

        builder.setMode(HttpMultipartMode.BROWSER_COMPATIBLE);
        builder.addPart("logfile", new FileBody(file));
        builder.addTextBody("filename", file.getName());

        try {
            post.setEntity(builder.build());
            HttpResponse response = client.execute(post);
            Log.d(TAG, EntityUtils.toString(response.getEntity()));

            if (response.getStatusLine().getStatusCode() == 201) {

                file.delete();
                Log.d(TAG, "Deleted file" + file.getName());

                return true;
            } else {
                return false;
            }

        } catch (ClientProtocolException e) {
            Log.e(TAG, e.toString());
            return false;

        } catch (IOException e) {
            Log.e(TAG, e.toString());
            return false;
        }

Pho3niX90, 2015-09-08:
Awesome thanks. Haven't been active in a while.

commit 6938f375 (2015-08-02): Use proper HTTP verb for uploading readings. Fixes #132