pires/android-obd-reader #135: crashes

sytolk, 2015-08-13:
I have so many app crashes. After select DTC after start live data with enabled GPS..
Can we include https://github.com/ACRA/acra to have device log?

Pho3niX90, 2015-09-15:
I have implemented something similar but in native android development. It creates a log file and asks the user if the log should be mailed.

pires, 2015-09-15:
@Pho3niX90 I am in favor of that solution instead of #138. Anyway, #138 is not ready for testing/merge.

sytolk, 2015-09-15:
ACRA support sending report by email too 
https://github.com/ACRA/acra/wiki/AdvancedUsage#sending-reports-by-email
You can put this preference in settings. But the main advantage that https://github.com/pires/android-obd-reader/pull/138 have its that its not need to have Email account (or SMTP) on Android device or Emulator. This is debugging feature (its only for the programmers and @pires can I know what is not ready for you ?) if you not like to debug application with my solution feels free to close PR and write new ACRA project for example ;)

Pho3niX90, 2015-09-15:
True. But acra is another library that needs to be taken into consideration when compiling. I agree that only developers needs logs, that's why you might as well use android native code and send a complete logcat to the developer. 

See here -> https://github.com/pires/android-obd-reader/pull/144

sytolk, 2015-09-15:
@pires you can think about debugging plugins for every kind of developer ;)

pires, 2015-09-15:
@sytolk my comments on why PR #138 is not ready for merge are detailed there.

closed 2016-05-12 by pires