pires/android-obd-reader #136: obd-java-api version 1.0-RC8 is broken

sytolk, 2015-08-31:
I have updated maven build to version 1.0-RC8 
https://github.com/pires/android-obd-reader/pull/133
But I think that obd-java-api lib in this build (v. 1.0-RC8) have some mixed java 1.7 and 1.8 issues.. maybe it depend of this issue:
https://github.com/pires/obd-java-api/issues/61
The error that I have is:
```
[INFO] --- android-maven-plugin:4.3.0:dex (default-dex) @ android-obd-reader ---
[INFO] Convert classes to Dex : /home/stanimir/AndroidProjects/android-obd-reader/target/classes.dex
[INFO] 
[INFO] UNEXPECTED TOP-LEVEL EXCEPTION:
[INFO] com.android.dx.cf.iface.ParseException: bad class file magic (cafebabe) or version (0034.0000)
[INFO] 	at com.android.dx.cf.direct.DirectClassFile.parse0(DirectClassFile.java:472)
[INFO] 	at com.android.dx.cf.direct.DirectClassFile.parse(DirectClassFile.java:406)
[INFO] 	at com.android.dx.cf.direct.DirectClassFile.parseToInterfacesIfNecessary(DirectClassFile.java:388)
[INFO] 	at com.android.dx.cf.direct.DirectClassFile.getMagic(DirectClassFile.java:251)
[INFO] 	at com.android.dx.command.dexer.Main.processClass(Main.java:704)
[INFO] 	at com.android.dx.command.dexer.Main.processFileBytes(Main.java:673)
[INFO] 	at com.android.dx.command.dexer.Main.access$300(Main.java:83)
[INFO] 	at com.android.dx.command.dexer.Main$1.processFileBytes(Main.java:602)
[INFO] 	at com.android.dx.cf.direct.ClassPathOpener.processArchive(ClassPathOpener.java:284)
[INFO] 	at com.android.dx.cf.direct.ClassPathOpener.processOne(ClassPathOpener.java:166)
[INFO] 	at com.android.dx.cf.direct.ClassPathOpener.process(ClassPathOpener.java:144)
[INFO] 	at com.android.dx.command.dexer.Main.processOne(Main.java:632)
[INFO] 	at com.android.dx.command.dexer.Main.processAllFiles(Main.java:510)
[INFO] 	at com.android.dx.command.dexer.Main.runMonoDex(Main.java:280)
[INFO] 	at com.android.dx.command.dexer.Main.run(Main.java:246)
[INFO] 	at com.android.dx.command.dexer.Main.main(Main.java:215)
[INFO] 	at com.android.dx.command.Main.main(Main.java:106)
[INFO] ...while parsing com/github/pires/obd/commands/control/DistanceTraveledSinceCodesClearedCommand.class
[INFO] 
[INFO] 1 error; aborting
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
```

If I use maven build of obd-java-api version 1.0-RC9-SNAPSHOT this problem not exist:
http://pastebin.com/1h0CCYyG
Maybe its better to use obd-java-api version 1.0-RC9-SNAPSHOT in gradle build too.

pires, 2015-08-31:
Yes. I'm taking care of this, thank you.

pires, 2015-08-31:
1.0-RC9 has been released with revert from Java 8 to 7 and a fix for the Windows platform.

sytolk, 2015-09-01:
Fix is for Linux platform too :+1: Thanks!

commit b097fdde (2015-09-01): Merge pull request #137 from sytolk/master

Upgraded to obd-java-api 1.0-RC9. Fixes #136
closed 2015-09-01 by pires