pires/android-obd-reader #139: Crash

sytolk, 2015-09-09:
```
09-09 16:44:26.514 D/com.github.pires.obd.reader.io.AbstractGatewayService( 5387): Adding job[29] to queue..
09-09 16:44:26.514 D/com.github.pires.obd.reader.io.AbstractGatewayService( 5387): Job queued successfully.
09-09 16:44:26.514 D/com.github.pires.obd.reader.io.AbstractGatewayService( 5387): Adding job[30] to queue..
09-09 16:44:26.514 D/com.github.pires.obd.reader.io.AbstractGatewayService( 5387): Job queued successfully.
09-09 16:44:26.515 E/ACRA    ( 5387): ACRA caught a StringIndexOutOfBoundsException for com.github.pires.obd.reader
09-09 16:44:26.515 E/ACRA    ( 5387): java.lang.StringIndexOutOfBoundsException: length=4; regionStart=0; regionLength=7
09-09 16:44:26.515 E/ACRA    ( 5387): 	at java.lang.String.startEndAndLength(String.java:583)
09-09 16:44:26.515 E/ACRA    ( 5387): 	at java.lang.String.substring(String.java:1468)
09-09 16:44:26.515 E/ACRA    ( 5387): 	at com.github.pires.obd.reader.activity.MainActivity$2.run(MainActivity.java:159)
09-09 16:44:26.515 E/ACRA    ( 5387): 	at android.os.Handler.handleCallback(Handler.java:800)
09-09 16:44:26.515 E/ACRA    ( 5387): 	at android.os.Handler.dispatchMessage(Handler.java:100)
09-09 16:44:26.515 E/ACRA    ( 5387): 	at android.os.Looper.loop(Looper.java:194)
09-09 16:44:26.515 E/ACRA    ( 5387): 	at android.app.ActivityThread.main(ActivityThread.java:5435)
09-09 16:44:26.515 E/ACRA    ( 5387): 	at java.lang.reflect.Method.invokeNative(Native Method)
09-09 16:44:26.515 E/ACRA    ( 5387): 	at java.lang.reflect.Method.invoke(Method.java:525)
09-09 16:44:26.515 E/ACRA    ( 5387): 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:857)
09-09 16:44:26.515 E/ACRA    ( 5387): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:624)
09-09 16:44:26.515 E/ACRA    ( 5387): 	at dalvik.system.NativeStart.main(Native Method)
09-09 16:44:26.515 D/ACRA    ( 5387): Building report
```

sytolk, 2015-09-09:
MainActivity.java
Line:159  sb.append(String.valueOf(mLastLocation.getAltitude()).substring(0, posLen));

posLen=7
remove this substing() for the Altitude

pires, 2015-09-09:
Please, provide PR :)

sytolk, 2015-09-09:
Ok but my previous PR  is not merged yet https://github.com/pires/android-obd-reader/pull/138 and these changes will be added to it ;)

pires, 2015-09-09:
@sytolk please don't do it that way or I won't be able to validate the PR and merge.

commit 628d6fe8 (2015-09-14): Fixes #139.
closed 2015-09-14 by pires