pires/android-obd-reader #141: Broadcast received BT messages

Pho3niX90, 2015-09-14:
Hi.

So I have been thinking of creating more activities that actually have some nice graphs and gauges. However, the current activity is using a handler, could we move this to a BroadccastReceiver so that all activities receive the same data. 

Or perhaps you know of a better way?

pires, 2015-09-14:
@Pho3niX90 the idea of broadcasting command results is interesting and my only recommendation is to use [LocalBroadcastManager](http://developer.android.com/reference/android/support/v4/content/LocalBroadcastManager.html).

Pho3niX90, 2015-09-15:
I agree. Although, not having a local broadcoast could allow for future plugin development? To keep the core app small and individuals install plugins?

pires, 2015-09-15:
The app is small enough for us to think about plug-ins.

Pho3niX90, 2015-09-15:
Okay I have implemented this in my app. One issue though. 

ObdCommand.java needs to implement Parcelable.

This is so that all activities receive the full package to call the methods as required. A workaround that I can think of now is to only send the formatted result which isn't ideal.

pires, 2015-09-15:
OBD Java API is pure Java so there's no way it will accept Android code.

Can't we wrap `ObdCommand`result in another object specific to this project that implements `Parcelable`?

Pho3niX90, 2015-09-15:
@pires that will be a better solution. I will have a look into it.

Pho3niX90, 2015-10-06:
Hi. 

I can't continue with this one. My project has dramatically changed and submitting PR's from it will just be catastrophic.

This is also a fairly simple change that I have already implemented in my project. However, I am not sure how you will get around not editing the java, maybe you should write a separate version of ObdCommand.java for this project?

I have to say this is the best change that can be made. After doing this I was able to bring my project to live by adding extra activities and having gauges in some of them.

pires, 2015-10-06:
I'll close this since you were the one suggesting it but no one has provided feedback or the will to pick it up.

Good luck with your project. Thank you for all your contributions.

closed 2015-10-06 by pires