pires/android-obd-reader #142: Issue ATS0 to remove spaces instead of programmatically removing them

Pho3niX90, 2015-09-15:
The Subject says it all. I think this will speed up processing commands, and remove unnecessary code.

pires, 2015-09-15:
@Pho3niX90 care to open a PR addressing this issue? Glad to review it.

Pho3niX90, 2015-09-15:
Sure stuff. Just need to review and test first, I have already added the spaces command to my project, still need to remove the redundant code.

Pho3niX90, 2015-09-15:
Need this first: https://github.com/pires/obd-java-api/pull/68

Pho3niX90, 2015-10-06:
Hi. 

I can't continue with this one. My project has dramatically changed and submitting PR's from it will just be catastrophic. However this change is very simple and will see if I can get some time to do it. For now I am closing.

closed 2015-10-06 by Pho3niX90

pires, 2015-10-06:
The thing with `AT S0` is that I can guarantee it to work with all ELM devices or that all programmers will send said command. But I can guarantee I remove all spaces, if any.

Pho3niX90, 2015-10-06:
That makes complete sense.