pires/android-obd-reader #152: Can't connect from HTC One

isgarsi, 2015-11-25:
Hi Pires,

I can connect with a mini ELM327 from several smartphones but I can't from a HTC One. It seems that there isn't any problem while It is connecting but when I try to get data from bluetooth socket, it returns a bluetooth exception (break pipe)

Could you help me with that?

Thanks in advance.

pires, 2015-11-25:
Probably the phone is broken. There's little to nothing I can do to help here, I'm sorry.

isgarsi, 2015-11-25:
I don't think so, because It can connect with other apps like torque. I'm going to try with your app android-obd-reader and will tell you the result.

Thanks

isgarsi, 2015-11-26:
It didn't run with android-obd-reader app, but it runs in all other smartphones so I won't waste more time with that.

Thanks.

closed 2015-11-26 by isgarsi