pires/android-obd-reader #157: Some missing DTCs

mvglasow, 2015-12-21:
I came across a few missing DTCs you might want to add:

    P0200 Fuel Injector
    P2181 Performance Malfunction in Cooling System
    P2187 System too Lean at Idle Bank 1
    P310B Fuel Pressure Outside Specification - Intermittent

I see that the guys over at OpenOBD (https://sourceforge.net/projects/openobd/) have a nice Sqlite database of DTCs and PIDs, a lot of which are not in Android OBD-II Reader yet. However, they're under GPLv3, so using their material would require changing the license of Android OBD-II Reader to GPLv3.

pires, 2015-12-21:
You need to follow GPL if you import or copy their code. There's no need for that. Thank you for the link.

Pho3niX90, 2015-12-22:
Here is the full list, http://www.obd-codes.com/trouble_codes/
However, most DTC are brand specific and descriptions change.

mvglasow, 2015-12-22:
@Pho3niX90 This is a really great resource! Unfortunately the bottom of the page states that all information on the page is strictly copyrighted, so we probably won't be able to just siphon off their data and feed it into the tool. They do permit linking, though – we could have each DTC open a link to their site (`http://www.obd-codes.com/$DTC`, replacing `$DTC` with the respective code).

As for brand-specific DTCs: obd-codes.com states that, of the P codes, only those beginning with P1 are manufacturer-specific. The same seems to be true for the C codes (the C0 codes appear to be generic, and they have a list of GM-specific codes starting with C1) and also of the U codes. The have no information on manufacturer-specific B codes, and all the codes they have start with B0. I'd suspect that, as with the others, any manufacturer-specific B codes would also start with B1.

The OpenOBD db has `manufacturer` and `model` fields but currently they are all set to "Generic".

About descriptions changing – different implementations may use different wordings, as in:

    P0441 Tank Venting System Function (Android OBD-II Reader)
    P0441 Evaporative Emission System Incorrect Purge Flow (OpenOBD)
    P0441 Evaporative Emission Control System Incorrect Purge Flow (obd-codes.com)

However, these are all referring to the same thing, with some descriptions being a bit more specific than others. But I'd expect the meaning of the codes to be stable – at least for the generic codes (other than B1, C1, P1, U1). Or do you have examples where the meaning changed?

Pho3niX90, 2015-12-23:
@mvglasow That is an interesting read and never knew that. I did have reference to an opensource db somewhere. I know we have discussed this in another issue somewhere, but since the trouble codes are law they are very generic and should follow a principle and guideline.

Here are more: http://www.nology.com/OBD2FaultCodes.htm
And another without any copyright notice: http://www.dtcsearch.com/

Pho3niX90, 2015-12-23:
I am busy populating a DB. Question is should this be kept in an Array (For easy translation). Or a sqlite db that cannot be translated.

pires, 2015-12-23:
Guess it depends on the size of the array.

Pho3niX90, 2015-12-23:
It will be 3 arrays, the key (code), the description and the make. Each being 7919rows

closed 2015-12-23 by pires

mvglasow, 2015-12-23:
@Pho3niX90 If it's not too late: a Sqlite DB could be translated as well if we included an extra column for language. The search key would then be DTC, make and language. The language field could be `null` for English descriptions, with some logic to fall back to that if no description is found in the current UI language.

It would take some extra coding: we'd initially need to determine the UI language. Then, on each lookup, we'd look for entries in that language, and if no result is found, repeat the search for the default language.

On the long run, I think a database is easier to maintain than a set of arrays.

Pho3niX90, 2015-12-23:
@mvglasow I also prefer a DB. But realistically people that will be translating will most probably not have the apps to open up a db and edit it.  On commits you also will not have the ability to see what actually changed. And the extra programming might be a pita and will require a lot of testing to make sure it doesn't cause issues with special characters.

PR #161 (2015-12-23): Added all possible DTC names to arrays.xml so to enable translation.
Fixes #157
Fixes #160
commit c48d232e (2015-12-23): Added all possible DTC names to arrays.xml so to enable translation.
Fixes #157
Fixes #160