pires/android-obd-reader #164: For PID:01 5E name: "ConsumptionRateCommand.class" there is Exceptions trouble

Kolyall, 2016-01-05:
For command: PID:01 5E name: "ConsumptionRateCommand.class"
Error:
W/System.err: java.lang.IndexOutOfBoundsException: Invalid index 3, size is 3
W/System.err:     at java.util.ArrayList.throwIndexOutOfBoundsException(ArrayList.java:255)
W/System.err:     at java.util.ArrayList.get(ArrayList.java:308)
W/System.err:     at com.github.pires.obd.commands.fuel.ConsumptionRateCommand.performCalculations(ConsumptionRateCommand.java:36)
W/System.err:     at com.github.pires.obd.commands.ObdCommand.readResult(ObdCommand.java:125)
W/System.err:     at com.github.pires.obd.commands.ObdCommand.run(ObdCommand.java:77)
W/System.err:     at com.github.pires.obd.reader.io.ObdGatewayService.executeQueue(ObdGatewayService.java:199)
W/System.err:     at com.github.pires.obd.reader.io.AbstractGatewayService$1.run(AbstractGatewayService.java:39)

Kolyall, 2016-01-05:
The same issue for this PIDs:

Pid:01 5E : ConsumptionRateCommand.class
Pid:01 34 : WidebandAirFuelRatioCommand.class
Pid:01 23 : FuelRailPressureCommand.class
Pid:01 21 : DistanceMILOnCommand.class
Pid:01 31 : DistanceSinceCCCommand.class

pires, 2016-01-05:
`performCalculations` is the culprit here.

```java
protected void performCalculations() {
    // ignore first two bytes [hh hh] of the response
    fuelRate = (buffer.get(2) * 256 + buffer.get(3)) * 0.05f;
}
```

Thing is, why are there only 3 bytes when there should be 4? Could this be a protocol thing? I call in the master of multi-protocol support, @valens254.

Kolyall, 2016-01-05:
@pires Maybe you should check size of `buffer`, without hardcode getting  `buffer.get(2) ` and `buffer.get(3)` ?

pires, 2016-01-05:
@Kolyall I could do that, but then the result would need to be an exception since the standard is pretty adamant about the result calculations. In this case, with my tests - and it's working for others - one would need to ignore the first two bytes of the response, namely `41 5E` which means `response to 01 5E`.

Kolyall, 2016-01-05:
About ignore of first two bytes I understand, and it is good for responce `41041B70`- after cut `1B70` is remaining -  The decoded number is `7024`

but if responce is `4104EA` - after cut `EA` is remaining -  The decoded number is  `234` - and here is bug reproduced

pires, 2016-01-05:
I'm not sure it _can_ be a single byte response, but in case you're right it should be easy to implement. Want to open a PR in `obd-java-api` which fixes this? Please, don't forget to add unit-tests that assert your code works.

Pho3niX90, 2016-05-11:
@Kolyall Can you please instead of using the calculated results, publish the rawData respons? This will explain what is going on

closed 2016-05-18 by pires