pires/android-obd-reader #167: For PID:01 1F name: "RuntimeCommand.class" there is Exceptions trouble

Kolyall, 2016-01-05:
For command: PID:`01 1F` name: "`RuntimeCommand.class`" 
com.github.pires.obd.exceptions.MisunderstoodCommandException: Error running 01 1F, response: ?
W/System.err:     at java.lang.reflect.Constructor.newInstance(Native Method)
W/System.err:     at java.lang.Class.newInstance(Class.java:1689)
W/System.err:     at com.github.pires.obd.commands.ObdCommand.checkForErrors(ObdCommand.java:203)
W/System.err:     at com.github.pires.obd.commands.ObdCommand.readResult(ObdCommand.java:123)
W/System.err:     at com.github.pires.obd.commands.ObdCommand.run(ObdCommand.java:77)

Kolyall, 2016-01-05:
The same issue for `01 2F` : `FuelLevelCommand.class`

pires, 2016-01-05:
I can't understand what's going on with this output. Care to disclose the full stacktrace?

Kolyall, 2016-01-05:
@pires It's full log stacktrace.

pires, 2016-01-05:
Commands may return exceptions. You need to handle them. In this case, the error seems to be related to the response to `01 1F` which seems to be wrong.

Since you're having so many issues, I'd say the protocol your car uses is mostly unsupported by `obd-java-api` library.

@valens254 seems to be the multi-protocol expert around and he's probably the most fit to help here again.

pires, 2016-05-17:
@Pho3niX90 would this be a CAN thing too?

Pho3niX90, 2016-05-17:
I think this is purely a protocol that is not supported by the vehicle. 
@Kolyall Which vehicle are you testing on and which year model? If I am correct this is not crashing the app, but just throwing the stacktrace? Thus far I have only seen 1 car that supports fuel level, and non that supports run-time (accurately).

closed 2016-07-16 by pires