pires/android-obd-reader #168: Error: java.lang.StringIndexOutOfBoundsException: length=11; regionStart=9; regionLength=3

Kolyall, 2016-01-05:
For PID: `03` : `TroubleCodesCommand.class`

W/System.err: java.lang.StringIndexOutOfBoundsException: length=11; regionStart=9; regionLength=3
W/System.err:     at java.lang.String.startEndAndLength(String.java:504)
W/System.err:     at java.lang.String.substring(String.java:1333)
W/System.err:     at com.github.pires.obd.commands.control.TroubleCodesCommand.performCalculations(TroubleCodesCommand.java:73)
W/System.err:     at com.github.pires.obd.commands.ObdCommand.readResult(ObdCommand.java:125)
W/System.err:     at com.github.pires.obd.commands.ObdCommand.run(ObdCommand.java:77)
W/System.err:     at com.github.pires.obd.reader.io.ObdGatewayService.executeQueue(ObdGatewayService.java:199)
W/System.err:     at com.github.pires.obd.reader.io.AbstractGatewayService$1.run(AbstractGatewayService.java:39)

pires, 2016-01-05:
@valens254 again, can you help? This is something you've worked on.

valens254, 2016-01-06:
I will look and check.

valens254, 2016-01-06:
@Kolyall I look and it seems like one frame with two DTC. But it message length must be 12 however into log "W/System.err: java.lang.StringIndexOutOfBoundsException: length=11;" 11. It is strange. May you check getResult() method?

closed 2016-03-28 by pires