pires/android-obd-reader #169: MQTT interface?

mstormi, 2016-01-24:
Hi,
thanks for your work!
Please let me frankly ask if you would be willing to add a MQTT server side interface?
Then everybody could run his own MQTT server (such as mosquitto running on a Pi at home, or rent a cloud MQTT service). There's solutions available to track, store and analyze MQTT-sent data that could be reused instead of running your own server, and even more important, it'll enable for all kinds of integration into 'the' Internet of Things. I'd love to get access to my car data in my openHAB smart home central (it has a MQTT interface already, as do other IoT applications).

best regards,
Markus

pires, 2016-01-26:
A server side interface? You mean, implement MQTT client that publishes data to one MQTT broker?

mstormi, 2016-01-26:
exactly

pires, 2016-01-26:
Is that a server-side interface? Anyway, seems interesting and I'm open to review a PR.

mstormi, 2016-01-27:
Oh, seems you got me a little bit wrong there. I don't have anything to contribute, unfortunately ... actually I meant this to be a feature request to you :-) Sorry for being unclear.
Btw, for a blueprint, you can check out the 'owntracks' app on Github.

closed 2016-07-16 by pires