pires/android-obd-reader #174: VIN - Binary result returned

KirbyJames, 2016-03-29:
Hi, I've rebuilt android-obd-reader-2.0-RC3 which I think contains obd-java-api-master 1.0-RC15 or RC14.

After looking at earlier VIN related prblems I've added ATH0 to the initialisation string.  However I still get (a fixed) binary value displayed for the VIN (except on about 1 in 10 occsaions when it does display the correct value).

I'd be grateful for any pointers as to what I might be doing wrong.

[image]

Pho3niX90, 2016-05-09:
This is still an issue on my side as well. I am prepared to put up a $10 reward for a proper fix for the VIN.

shareef593, 2016-05-09:
Add $10 from me

Pho3niX90, 2016-05-10:
For all that are having this issue, please post your rawData vin here. I need to see what vins are being returned so that I can figure out one algorithm.

KirbyJames, 2016-05-11:
Many thanks for returning to this issue.

Below are two VINs shown as both raw and hex

    euÂ¥Â¥Â£ÂµÂ¤ #3s#                               
    eÂ…Â„Ã”Â£ £ÂƒSs 

[image]

Pho3niX90, 2016-05-11:
@kirbyjames
Could you please post your raw response from the obd. So that I can do tests?

Pho3niX90, 2016-05-11:
@KirbyJames Pleaes try this project: http://goo.gl/uAcCK3
Replace the result string with your raw response. See if it's giving you the correct output

KirbyJames, 2016-05-11:
Hi, I've added the code to VinCommand.java and rebuilt the app. Unfortunately it returns the same value ..

[image]

Pho3niX90, 2016-05-11:
@KirbyJames I will need your rawData response to rrsolve.

Pho3niX90, 2016-05-11:
in calculated results. Instead of returning vin return tawData

Pho3niX90, 2016-05-11:
sry I am in the phone. mean rawData

Pho3niX90, 2016-05-11:
@KirbyJames 
Please add the following in you vin data
`Log.d("VinCommand.java", "rawdata: " + rawData);`

Like so:
[image]

Then find the raw response in your logcat.

KirbyJames, 2016-05-11:
Ah. I have realised that my build is using the github version of obd-java-api-master/.  I have downloaded the obd-java-api-master library - but I don't know how to modify the gradle configuration to use it in the build.

Pho3niX90, 2016-05-11:
@KirbyJames I have no idea to be honest. I have completely moved away from gradle. 
@Pires do you know the answer to the above question?

KirbyJames, 2016-05-11:
I've managed to re-build the app with the modified VinCommand.java
I've printed the rawdata value below ... and a valid 17-character Vin is returned :-)

rawdata: 0:4902015756571:5A5A5A314B5A422:57323233373237
    Vin: WVWZZZ1KZBW223727

rawdata: 0:4902015756571:5A5A5A314B5A422:57323233373237
    Vin: WVWZZZ1KZBW223727

Pho3niX90, 2016-05-11:
@KirbyJames Awesome news! What vehicle was it used in, which country and what year model is the vehicle? This will help greatly in diagnosing further enchancements to the VIN.

KirbyJames, 2016-05-11:
WVWZZZ1KZBW223727 is a 2011 VW Golf registered in the UK.
Will the update be included in an updated obd-java-api?

Pho3niX90, 2016-05-11:
Then it's most def a can. on same restriction as we here in south africa are, except no emissions. 
Anyways, yes the changes will be added to the lib, pullrequest is located here -> https://github.com/pires/obd-java-api/pull/113

pires, 2016-05-12:
I'll be releasing library RC16 as soon as @Pho3niX90 finishes an open PR and will update this project.

pires, 2016-05-17:
Done.

closed 2016-05-17 by pires