pires/android-obd-reader #183: When service is running and you open the notification, it fails to check if the service is running

phamels, 2016-05-27:
Hi,

I've noticed that when you start the service, the notification is created and the service is actually running, when you click on the notification to return to the main activity, it fails to check if the service is actually running even though it is running.

So if (service != null && service.isRunning()) always returns a false even though it is still running in the background but the main activity thinks it's not and behaves that way, resulting in no communication between the service and activity anymore.

pires, 2016-05-29:
@phamels thanks for reporting. Care to open a PR that fixes this?

pires, 2016-07-07:
@jbits want to check this one out? ;-)

jbits, 2016-07-07:
I can't reproduce it. From what I understand after I go back to the app by clicking on notification (service != null && service.isRunning()) evaluates to false. I don't see it happening. Maybe somebody could explain more clearly what is the issue. I'm fairly new to android.

closed 2016-07-16 by pires