pires/android-obd-reader #186: Update data too much delay + Can not get Distance traveled

kaorijp, 2016-06-15:
Hi 
I was using ODB reader to test my app in real Car.
And I have some confusion with current source code.

1. The Speed data update is too later than real.
2. Can not get Distance Traveled.

Please help me to confirm 
Thanks

Pho3niX90, 2016-06-15:
1. This is because OBD2 protocol is not lightning fast, there will always be a delay based on the ECU of your vehicle + OBD2 adapter being used. 

2. As for distance traveled, is your GPS working?

kaorijp, 2016-06-15:
Thank for your reply. 
I would like to confirm as below

1. I was testing with adapter OBD II Multi Meter that was supported by OBD Info-san MAXWIN application.
It's working fine
- Updating is  as fast as real kilometer on car. ( I feel it is no delay and very smooth )
So, when I switch to use OBD-reader, the updating is very late. 

2. I removed the GPS function on app ( by customize source code )
Because , OBD 2 is supporting command to get Distance , right ?
If I use GPS, so OBD adapter becomes meaningless

kaorijp, 2016-06-15:
In addition, I can get distance by using OBD Info-san MAXWIN without turn on GPS.
I confirmed that

Pho3niX90, 2016-06-15:
@kaorijp if you do not like that it uses GPS to get accurate distance traveled as opposed to a OBD2 PID that is unreliable then I suggest you modify it to your liking. 

As for speed issues, the app by default polls ALL sensors and will slow down individual PIDs. So turn off all sensors that you are not interested in.  The app you mentioned does not poll more than 3 to 4 sensors at a time I assume.

kaorijp, 2016-06-15:
@Pho3niX90 
Thank for your suggestion. 
I'll try to do that

pires, 2016-06-15:
@Pho3niX90 always to the rescue, thanks man!

closed 2016-06-15 by pires

sytolk, 2016-06-15:
@pires @Pho3niX90 do its have command implemented in obd-reader that can return vehicle KM traveled? Its will be good to match it with the GPS distance.

kaorijp, 2016-06-15:
> @pires @Pho3niX90 do its have command implemented in obd-reader that can return vehicle KM traveled? Its will be good to match it with the GPS distance.

This is my confusion. I try to confirm that.
Theoretically, we can do that.

Pho3niX90, 2016-06-15:
@sytolk It's available, but I have never gotten anything reliable or remotely accurate from it. The thing is all mobile devices today have a proper built in GPS to properly calculate the distance traveled. Being a business intelligence developer by trade I always opt for the most accurate data possible. 

@pires always a pleasure

deft5746, 2017-10-27:
I have a ELM327 device and i want to calculate distance from start point to end point with ELM327 device and not with GPS.I am a newbie and I am unable to find anything on the internet. Can anyone help with this? Any help would be appreciated.

tamafuda, 2017-10-27:
@deft5746 Are you working on Simulator or Real Car ?
There are a lot of car models with difference protocols among difference manufactures.
With some new model car we can get distance without GPS . But of course it depends on car models ( support or not support )

deft5746, 2017-10-27:
Thank you for replying,
We were working on a simulator as we are just setting up the code right now but we didn't get the code for calculating the distance. Once we are done with the code setup then we will try it on a real car. We were working with this till now :

[https://github.com/Hesamedin/ELM327](url)