pires/android-obd-reader #187: ELM327 OBD II problem

kaorijp, 2016-06-20:
Hi 

I was testing with ELM 327 OBD II adapter but there are many problem with it.
Here is the adapter:
https://www.amazon.co.jp/gp/product/B00J7CC4T2/ref=s9_simh_gw_g263_i3_r?pf_rd_m=AN1VRQENFRJN5&pf_rd_s=desktop-1&pf_rd_r=CWQZA5C1423S3A1CGSS4&pf_rd_t=36701&pf_rd_p=9d539fa0-72bc-4756-8b94-2163f0a38a88&pf_rd_i=desktop

After connect to adapter, all of initial value is very strange. 
Speed is 11565 ( It should be 0 km/h)
RPM, Distance v.v.v.. also is really strange.

Has anyone seen this problem before ?
Thank

bitaromulus, 2016-06-20:
Hello,

I am also receiving some strange values for most of the parameters. I am using the ECUsim 2000 simulator and for example, the Speed and RPM is displayed like this:
Speed: 410D941:DAF11803410D942:DAF12803410D94
RPM: 410CFFFF1:F11804410CFFFF

When i manually change the value of the parameters of the ECUsim, these parameters are updated but are displayed in the same format.
Is this a format issue, incompatibility or am i doing something wrong ?

Thank you !

Pho3niX90, 2016-06-20:
@bitaromulus that is incompatibility. You are seeing can responses. I am busy working on getting can to work, however my vehicle's do not respond in can so it's taking longer than expected. 

Could you please post more raw responses along with what the value should be (roughly).

Pho3niX90, 2016-06-20:
@kaorijp could you please post your raw responses?

bitaromulus, 2016-06-20:
@Pho3niX90 

Hello again,

My ECUsim 2000 has 5 knobs which can be used to manually update the values of the parameters.
The parameters that can be monitored are:
- Knob 1: Engine Coolant Temperature;
- Knob 2: Engine Speed (RPM);
- Knob 3: Vehicle Speed;
- Knob 4: Oxygen Sensor Voltage (I don't know if this is actually available in the current API, didn't check);
- Knob 5: Mass Air Flow (MAF).

This simulator, also has a Configuration DIP Switch (has two switches that set the protocol attributes for ISO 15765-4(CAN). Switch number 1 is also used to select the init type for ISO 14230-4: 
a) 29 bit / 11 bit - Selects CAN frame ID type;
b) 500 kbps / 250 kbps - Selects CAN baud rate;
c) Fast Init / 5 Baud Init - Selects the type of initialization for ISO 14230-4).

When i wrote the previous message, i was using the simulator with the following settings: 
29 bit / Fast Init + 500kbps
With these settings, i was receiving those strange values for most of the parameters, to be more precise: Knob 1, Knob 2 and Knob 3.

Meanwhile, i changed to the following settings:
11 bit / 5 Baud Init + either 500 kbps or 250 kbps
After i did this, i ran the Android application once again and i've noticed that the values for Knob 1(Engine Coolant Temperature) and Knob 2(Engine Speed (RPM)) are displayed correctly. Knob 3(Vehicle Speed) is still the same strange format, though.

Currently, for Knob 3(Vehicle Speed) the displayed value is as follows:
- With Knob 3 set at max, the displayed value is: 410DFF2:0DFF4000103700
- With Knob 3 set at min, the displayed value is: 410D002:0D003000103700

Knob 3 should give as a returned value 0x00 for min and 0xFF for max (0~255km/h).
What is suprisingly is that certain bits of the displayed parameter, are constantly updated:
410DFF2:0DFF**40**00103700

The numbers marked in bold are being updated constantly (it doesn't has an exact period of time, it just changes at about 5~10 seconds).

If you need any other information, please let me  know and i will try to help in whatever way is possible for me.

Thank you in advance !

kaorijp, 2016-06-21:
Good morning !

@Pho3niX90 
Regrading the raw data, I'm sorry that I didn't save it before. I will take the screenshot or file log in next testing time. 
BUT, all of strange value look likes a [Max value] that is described on Wiki page:
https://en.wikipedia.org/wiki/OBD-II_PIDs

For examples :
RPM 16,383
Distance 65,535
etc...

@bitaromulus 
Have you think the protocol type is a one of reasonable ?
( At that time, I choosed AUTO )

bitaromulus, 2016-06-21:
Good morning,

@kaorijp I've tried with all of the available protocols and only the CAN ones provide good data (except for the Vehicule Speed). The other ones do not work, NODATA is displayed in most of the cases.
Currently, i also use AUTO setting.
This evening, i will also try to post a screenshot and if possible, a log file. I guess this way, it will be better.

Thank you and have a nice day !

kaorijp, 2016-06-21:
@bitaromulus 

Thank for your reply.
As far as I know, the Vehicle speed is the most important value. 
Anyway, I'm waiting your next report.
I will also try to test with all protocols. I just need to get Timing, Speed, Distance only .
Report you soon....

bitaromulus, 2016-06-21:
Hello once again,

I've tested once again the application but it's the same problem with the vehicle speed.
I am attaching the log file, i hope it helps to identify more easily the source of this problem:
[OBDReader_logcat_1466540907052.txt](https://github.com/pires/android-obd-reader/files/326537/OBDReader_logcat_1466540907052.txt)

If you need any other information, please let me know.
 
Thank you and have a nice day !

kaorijp, 2016-06-22:
Hello @bitaromulus 
I have a question for you.
> I've tried with all of the available protocols and only the CAN ones provide good data (except for the Vehicule Speed).
What kind of CAN protocol that has been confirmed ?
There are many CAN protocols type that is defined in source code .
Ex: 
IOS_15765_4_CAN
IOS_15765_4_CAN_B
IOS_15765_4_CAN_C
IOS_15765_4_CAN_D
SAE_J1939_CAN
USER1_CAN

bitaromulus, 2016-06-22:
Hello @kaorijp 

You are right, i should have been more specific regarding the protocol which i use.
My ECUsim 2000 is not full featured, this means that it currently supports only the Standard ISO 15765-4.

I currently have chosen the AUTO option from the configuration menu, but it always uses IOS_15765_4_CAN_C (except for the Vehicle Speed, everything works for me with this protocol).
If i choose a different protocol, NODATA is received/displayed.

I hope this information is useful for you.
Have a nice evening !

kaorijp, 2016-06-23:
@bitaromulus

I'm planing to buy a OBD 2 Simulator such as ECUsim.
So I have been testing on real car ( Toyota Voxy 2014).
I didn't test with each of protocol type. Just set as AUTO.
Beside that, I do not use Wifi , my expected is what is the real result that retuning by OBD II command only .
I dont want to get distance by GPS. 
Can you help me to test Distance MIL with out GPS ?

Pho3niX90, 2016-06-23:
@kaorijp Distance MIL is not distance traveled, it's distance traveled with MIL light on.

Pho3niX90, 2016-06-23:
@bitaromulus thank you for the examples you have posted. 
My pc crashes and need to resetup everything from scratch again. Once I have done that I will work on a solution for the can messages.  I will look into acquiring the ecusim as well for the multiprotocol development. Currently I use win-obdsim for testing but only supports kline messages.

Pho3niX90, 2016-06-23:
@bitaromulus 
Could you please post example of these mods
29 bit / Fast Init + 500kbps
11 bit / 5 Baud Init + either 500 kbps or 250 kbps

With 2 raw examples of each knob, for each protocol above? That would be a considerable help to improve the library @ https://github.com/pires/obd-java-api/issues/115

kaorijp, 2016-06-23:
Good evening!
@Pho3niX90

Thanks for your explain.
So , OBD 2 is not currently supporting distance traveled command (PID) ?

Pho3niX90, 2016-06-23:
@kaorijp no it doesn't have such a function. Only distance travelled with MIL on, and since codes cleared. See https://en.wikipedia.org/wiki/OBD-II_PIDs

bitaromulus, 2016-06-23:
Hello @Pho3niX90 

I have managed to group some raw incoming/outgoing data for each of the Knobs.
In the pictures, you will see incoming/outgoing data from the ECUsim 2000, which i've captured using TeraTerm. I've started with min value for each knob and gradually increased it until the max value (you will notice that the values will modify).
Also, for each Knob, i've associated the log file.

**11 bit / 5 Baud Init + 250 kbps**
**Knob 1 (Engine Coolant Temperature)**
ECUsim LOG: 
[image]

Android App LOG: -  (I forgot to send the logfile. If it's necessary, i can upload it later.)

**Knob 2 (RPM)**
ECUsim LOG: 
[image]

Android App LOG: 
[[RPM]OBDReader_logcat_1466705645131.txt](https://github.com/pires/android-obd-reader/files/330633/RPM.OBDReader_logcat_1466705645131.txt)

**Knob 3 (Vehicle Speed)**
ECUsim LOG: 
[image]

Android App LOG: 
[[VH. SPEED]OBDReader_logcat_1466706265806.txt](https://github.com/pires/android-obd-reader/files/330634/VH.SPEED.OBDReader_logcat_1466706265806.txt)

_Knob 4 (Oxygen Sensor Voltage) - Not available_
_ECUsim LOG: -
Android App LOG: -_

**Knob 5 (Mass Air Flow)**
ECUsim LOG: 
[image]

Android App LOG: 
[[MAF]OBDReader_logcat_1466706011275.txt](https://github.com/pires/android-obd-reader/files/330635/MAF.OBDReader_logcat_1466706011275.txt)

**29 bit / Fast Init + 500kbps**
**Knob 1 (Engine Coolant Temperature)**
ECUsim LOG: 
[image]

Android App LOG: 
[[ENGINE COOLANT TEMP.]OBDReader_logcat_1466707547676.txt](https://github.com/pires/android-obd-reader/files/330638/ENGINE.COOLANT.TEMP.OBDReader_logcat_1466707547676.txt)

**Knob 2 (RPM)**
ECUsim LOG: 
[image]

Android App LOG: 
[[RPM]OBDReader_logcat_1466708093893.txt](https://github.com/pires/android-obd-reader/files/330640/RPM.OBDReader_logcat_1466708093893.txt)

**Knob 3 (Vehicle Speed)**
ECUsim LOG: 
[[VH. SPEED]OBDReader_logcat_1466707754477.txt](https://github.com/pires/android-obd-reader/files/330642/VH.SPEED.OBDReader_logcat_1466707754477.txt)

Android App LOG: 
[image]

_Knob 4 (Oxygen Sensor Voltage) - Not available_
_ECUsim LOG: -
Android App LOG: -_

**Knob 5 (Mass Air Flow)**
ECUsim LOG: 
[image]

Android App LOG: 
[[MAF]OBDReader_logcat_1466707977783.txt](https://github.com/pires/android-obd-reader/files/330643/MAF.OBDReader_logcat_1466707977783.txt)

When do you think it will be possible to fix the displayed format for the Vehicle Speed ?

If you need any more information, let me know and if it's possible for me, i will try to help.

Thank you and have a nice evening !

kaorijp, 2016-06-27:
@bitaromulus 
@Pho3niX90 
I'm so sorry for ask you again about Distance traveled.  
I'm really confused with three Module $ PID
- 01 21   -> Distance traveled with malfunction indicator lamp (MIL) on
- 01 31   -> Distance traveled since codes cleared
- 04 NA  -> Clear trouble codes / Malfunction indicator lamp (MIL) / Check engine light
(Clears all stored trouble codes and turns the MIL off.)

From my understanding,we can get value that return from command **Distance travelled with MIL on**
And then, if (04 NA) command has been called , it means the MIL if OFF.  At this time, we can use command (01 31) to get value in this case ( Since codes cleared) . What does value of (01 31) mean ?

So my confusion is :
- What is the difference beween **distance travelled** , **distance travelled with MIL on** and **distance travelled with MIL off** ( in case of since codes cleared)

Pho3niX90, 2016-06-27:
@kaorijp What are you trying to achieve?

kaorijp, 2016-06-27:
@Pho3niX90 
Ex: A car travels from point A to point B. 
How many km driven in this case?

Pho3niX90, 2016-06-27:
@kaorijp Then there is no pid for you. 31 has a max value it will display and does not get reset on each trip. 31 is distance travelled after the last time errors were cleared. And it's not supported by most vehicles as far as I have seen

sytolk, 2016-06-27:
@Pho3niX90 I have KM on the display of my car :) and I suppose that its have some pid to get this KM. If its have this KM (of production without reset) a car travels from point A to point B = km B - km A

Pho3niX90, 2016-06-27:
@sytolk There is no generic PID for this. Therefore not possible in this project. You will have yo run a sniffer on your vehicles network to find it. But this is different per brand and per model. I personally do not see a need for this so cannot assist in persuing it when the gps works flawlessly.

Pho3niX90, 2016-06-27:
@kaorijp My only suggestion to what you want to achieve is monitor the cars speed over the session time and calculate the distance.

sytolk, 2016-06-27:
@Pho3niX90 this is good idea. @kaorijp can you check distance calculated from speed like: List<Integer> speedChanges = [30,10,30,25,5];
if(curr_speed != previous_speed) //add to array
int avgSpeed = 100/5=20 km/h
distance (km)= 20 x time(h)

sytolk, 2016-06-27:
Its wrong :) its need to have timestamp attached to the HashMap speed => timestamp

HIGHLANDER3, 2016-06-27:
[image]

If you take the memorized distance traveled and subtract from the actual and display it, that could satisfy?
You just need to consider that other option when he uses the other PID for the calculation the distance traveled due to MIL ocurrence.
It will not be good if you clear those trouble codes, once you need it to know whats going on in your vehicle.

sytolk, 2016-06-27:
Maybe some mix if pid 31 and 21 supported by vehicle use it. If not it will be good to have distance based on speed changes or to enable both methods +GPS and match the results. Who can provide PR ? :)

kaorijp, 2016-06-28:
@Pho3niX90 
>31 has a max value it will display and does not get reset on each trip

What does max value mean in this case ?

@sytolk
@Pho3niX90 
>My only suggestion to what you want to achieve is monitor the cars speed over the session time and calculate the distance

Yep. This is my workaround until now.
I also tested this case and has a litle difference from the real kilometer on car.

### Results 
In fact, I have driven 2.2km and spent 15 minutes. But the calculation is returned 1.9 ~ 2.05 km 
Maybe the calculation the not good BUT the main problem is the speed is not updated in real time !
( Around 2 seconds is delay than real kilometer )

closed 2016-07-20 by hoangvm