pires/android-obd-reader #196: Car Control feature?

achinverma, 2016-09-01:
Hi, i have used this api in my car and it is running finr, but is it possible to send some commands so that we control some car features like, via sending any command from this api to car using obd-reader reader , 
i can trigger car horn or Head Lamps , please let me know if this is possible using this api or any other?

pires, 2016-09-01:
The API (obd-java-api) used in this project allows you to send any command you desire.

closed 2016-09-01 by pires

achinverma, 2016-09-01:
ok , i mean to say that i can turn on/off car Horn or Head Lamps via this api from android mobile app , please share any example or code ?

pires, 2016-09-01:
OBD-II PIDs are not meant for that. You must consult your car manufacturer to get the proprietary commands to execute.