pires/android-obd-reader #197: Wrong readings

boriskn, 2016-09-13:
I have wrong readins for data on that application. I'm getting fixed rpm value (4163) for example. Other readings come up as fixed also. 

Could you help me please.

pires, 2016-09-13:
You should try different protocols since auto selection isn't working.

boriskn, 2016-09-14:
I triead all the protocols...they all turn the same static values if they are successful. Else I get ERROR or NODATA. 

I tried with torque and it works well on no matter what protocol I use. 

I'm testing everything with Freematics OBD emulator if that helps. 

Do you have any idea what else could I try? 

Thanks!

pires, 2016-09-14:
It's probably CAN stuff that is not well supported right now.

boriskn, 2016-09-14:
What do you reccomend that I can do now to continue testing / developing the app?

jbits, 2016-09-15:
> The speed was displayed as a constant 65km/h while standing still and engine RPM of 4163.
Also the runtime was a constant 04:37:51.

The GUI is not always refreshing correctly when OBD status  is disconnected. Make sure it's connected before attaching any significance to the readings.

vinod123igloble, 2016-11-18:
Hi Pires,
             Thanks for the demo code its really helpful but i have some problem as i am not able to getting distance travelled and Fuel consumption,fuel level by this.can you please help me out of this.

thanks in advance