pires/android-obd-reader #203: get_data.php  sample

githubsrinath, 2017-01-10:
Is there any get_data.php sample program available to read and store the data in the centralized server.

If not... I will write my own get_data.php.  But I would like to know, what variables I need use in get_data.php in order to get the data from app?

Many Thanks
Srinath Rao

[image]

pires, 2017-01-10:
You can base your work in this https://github.com/pires/android-obd-reader/issues/147. But a github/google search would probably give you PHP code as well.

closed 2017-01-10 by pires

M-Ahmad-sattar, 2017-08-15:
sir is still any referral information regarding this point?